ExpressLRS/ExpressLRS #2909: HoTT Telemetry protocol synchronisation enhancement

icebreaker-ch, 2024-09-02:
## Current Behavior

HoTT Telemetry connections require a special (3 to 4 wire) cable, using a diode from RX to TX.
This is ok for "real" HoTT telemetry items.

I tried, to connect an Arduino board with full duplex (4 wires) to the receiver, supposing the diode is not required in this case, but it did not work [](https://github.com/icebreaker-ch/OpenHoTTSensor)

Studying the code in the file SerialHoTT_TLM.cpp, I found that synchronisation relies on the 2 byte echo (REQUEST_ID and MODULE_ID) of the Bus Master (Receiver):

```
void SerialHoTT_TLM::processBytes(uint8_t *bytes, u_int16_t size)
{
    hottInputBuffer.pushBytes(bytes, size);

    uint8_t bufferSize = hottInputBuffer.size();

    if (bufferSize == sizeof(hottBusFrame))
    {

```
I think, it is not necessary, to include the echo in the synchronisation process and therfore enable the possibility to use a 4 wire (full duplex) or 2 wire connection.

## Steps to Reproduce

- Connect a HoTT enabled receiver by a 4 wire (full duplex) connection
- Send HoTT data only (without echoing the request from the Bus Master)
- Receiver will not synchronise and no telemetry data is transmitted.

## Possible Solution (Not obligatory)
Change code in SerialHoTT_TLM.cpp to synchronize on Start of the plain HoTT Protocol (0x7C 0x8? ...)

## Your Environment

- TX hardware: FrSky X10S and RadioMaster ELRS Module
- RX hardware: BetaFPV ELRS Micro 5 Channel PWM Rx
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): 3.4.3
- Packet Rate: 100 Hz Full
- Telemetry Ratio: 1:4

mha1, 2024-09-02:
No can do.

Only EPS8285 receivers expect a diode between RX and TX to form the expected half duplex single wire serial physical layer because they can't do half duplex in hardware. EPS32 can do that hence ESP32 receivers only use TX as a true half duplex physical layer. Your solution is violating the HoTT protocol specification as it doesn't use a half duplex single wire so won't work in a mixed true HoTT sensors / you Arduino sensor scenario and won't work on ESP32 receivers at all.

Change your code to communicate over a half duplex serial connection. It's fairly easy to do using SoftSerial.

mha1, 2024-09-02:
Closed - working a specified and required

closed 2024-09-02 by mha1

mha1, 2024-09-15:
I answered to your discussion here: https://discord.com/channels/596350022191415318/1283765140080300062