ExpressLRS/ExpressLRS #2912: RX Failsafe Position when using serial link

Arielofek, 2024-09-03:
### Discussed in https://github.com/ExpressLRS/ExpressLRS/discussions/2911

<div type='discussions-op-text'>

<sup>Originally posted by **Arielofek** September  3, 2024</sup>
Hello,
In my set up I have an ELRS receiver linked through SBUS to a gyro with Level mode capability. I would like to set the failsafe position of the channel that controls the gyro mode to Level mode. However, the failsafe position set up of ELRS RX only works for PWM outputs. This forces me to give up the nice single control line of SBUS and go back using the old PWM setup which is less desired because of the additional wire connections. I will be more than happy to see in next ELRS releases an update that sets the failsafe position in both PWM and SBUS/other serial links.</div>

pitts-mo, 2024-09-04:
A typical device accepting serial RX these days should allow you to configure a fail-safe state that is triggered by the fail safe indicator of the serial RX. However, being able to set the state of each channel during a fail-safe event would be a nice option to have available for old/simple devices.

What is the specific gyro you are using?

Arielofek, 2024-09-04:
I'm using the Hobbyeagle A3S4. The failsafe within the gyro is set to Normal mode (wind rejection) and cannot be changed. My preferred failsafe mode is Level mode (panic mode) for having the best chance to survive with minimal damage. For that I need the receiver to failsafe into Level mode. I have a Happymodel EPW6 receiver. I prefer using serial link with the gyro but the failsafe pre-set position (through wifi) only works at the PWM outputs. At the serial link, LAST POSITION and NO PULSE are the only supported failsafe modes. Bottom line, we need this feature. I can't see the reason for this difference between PWM and the serial link. They are both outputs and should be treated the same way.

Ayil911, 2025-02-07:
Any update from Dev's on this issue?

Arielofek, 2025-12-19:
> Yes an option to set channel failsafe positions in serial mode would be a
> nice addition.
> e,g. Goosky FBL controllers (except rotorflight based) doesn't have custom
> failsafe position settings / also mainstream cheap chinese controllers do
> not have that option

Hi Edward,
I received this email with your response. I hope the ELRS team picks up our
comment and implements it in one of the next releases. By the way, somehow,
John Hopkin's channel on Youtube (RC Video Reviews) went silent. Do you
know what happened?

Thanks,
Ariel

briankoren, 2025-12-29:
Not talking about SBUS, no. I'm using CRSF. I think being able to set failsafe states per channel would be useful regardless of the protocol