ExpressLRS/ExpressLRS #2914: HM EP1 no wifi on 3.5-rc1

TommyV6, 2024-09-04:
Flasing slow, bind OK

Does not turn on WIFI at ALL.

Flash to 3.5-RC1

Sollution
Flash to pre 3.4 firmware.

## Details

In version 3.3.X, EP1 works correctly, after 60 seconds it starts flashing rapidly and activates Wi-Fi. In version 3.4.X, EP1 flashes rapidly twice after 60 seconds and then returns to slow flashing without activating Wi-Fi. In version 3.5-RC1, EP1 keeps flashing slowly and doesn't activate Wi-Fi at all. In version 3.5-RC1, the moment of restart (rapid flashing) is also not visible. The receiver binds correctly in version 3.5-RC1. Flashed with uart with full erase.

It looks like fix: https://github.com/ExpressLRS/ExpressLRS/pull/2848 does not work ?

## Your Environment

- TX hardware: RM Boxer
- RX hardware: HM EP1 old one
- Handset model: RM Boxer
- OpenTX version latest EdgeTx
- ExpressLRS version (TX & RX MUST MATCH): yes
- Packet Rate: 150hz
- Telemetry Ratio: none
- user_defines: only password

pkendall64, 2024-09-09:
> Bug was fixed in 3.5-rc2 #2897 now everything works.

I find this very odd considering there we zero bug fixes related to WiFi on RC2, and the PR you are referring to (#2897) was actually in RC1!

TommyV6, 2024-09-10:
My EP1 does not turn on wifi on RC1. I got two HappyModel EP1`s.

closed 2024-09-10 by TommyV6

sashaegorov, 2024-10-14:
@TommyV6 hey got the same issue.
Have you fixed this or just rollback to version 3.3.x

TommyV6, 2024-10-15:
> @TommyV6 hey got the same issue. Have you fixed this or just rollback to version 3.3.x

Issue was fixed in 4.5.0 but on my EP1 wifi stopped working, after 4.5.1 everything is working like it should.