ExpressLRS/ExpressLRS #2920: Telemetry is frequently lost and recovered

chofchop, 2024-09-07:
## Current Behavior
Telemetry is frequently lost and recovered

## Steps to Reproduce
See video.

## Possible Solution (Not obligatory)

## Details
This problem has been occurring since v3.4. It did not occur up to v3.3.
Also, it does not occur with the external module (BetaFPV NanoTX), so it is probably a problem specific to RX_as_TX.

As you can see in the video, the call says "telemetry lost," but it doesn't seem like telemetry has actually been lost. No control lost has occurred either.

## Your Environment

- TX hardware: EP1tcxo(RX_as_TX)
- RX hardware: EPW6tcxo
- Handset model: MT12
- EdgeTX version (including nightly number) : 2.10.3
- ExpressLRS version (TX & RX MUST MATCH): 3.5.0 RC2
- Packet Rate: F500
- Telemetry Ratio: 1:32
- PWM frequency: 333Hz,333Hz,160Hz,50Hz,50Hz,50Hz

https://github.com/user-attachments/assets/fdeaa737-6ad7-4bf6-a124-5c95500e7652

mha1, 2024-09-07:
what baud rate are you running?

chofchop, 2024-09-08:
I basically use 3.75MHz, but I also tried 1.78MHz and 5.25MHz, but there was no improvement.

0crap, 2024-09-08:
Same here, including a full link loss, which is even worse. (v3.5 RC2)
Nothing special like the OP with RX_as_TX.
Just simple setup with LoRa mode that works without a glitch on v3.4.3
I'll make my own issue with full detail when I have some more time.

pkendall64, 2024-09-09:
Just hooked up a BetaFPV Lite RX, flashed as a TX (3.5.0-RC2) using the new web-flasher (https://pkendall64.github.io/web-flasher/).
Connected to T20 as an internal module (because it's an 8285 receiver).
Running at 1.87M baud, limited by T20. Factory firmware (67f978a3). No issues after 10 minutes.
Flashed with EdgeTX 2.10.2, No issues for 10 minutes, then a couple of callouts then all good again.

Can you provide a photo of the connection to the MT12 and any configuration required. I can try it on mine as well.
I would suggest reflashing with the erase option set so it fully erases any data on the receiver.
Also, if possible, try with an ESP32 based receiver as that should be better supported as the TX code is much more tested on that platform.

chofchop, 2024-09-09:
[image]
[image]
[image]

gagarinlg, 2024-09-09:
Just increase the UART speed for the internal module.

chofchop, 2024-09-09:
I am flashing and trying an EP1/2 tcxo and a BetaFPV nano receiver as TX. Unfortunately I don't have an ESP32 receiver.
As mentioned before, I get frequent "telemetry lost, recovered" calls but no connection lost. That being said, I don't think it's a problem that can be ignored.

Further test results.
TX:EP1tcxo-v3.5.0, RX:EPW6-v3.5.0 > NG
TX:EP1tcxo-v3.5.0, RX:EPW6-v3.3.2 > NG
TX:EP1tcxo-v3.3.2, RX:EPW6-v3.5.0 > OK
TX:BetaFPVnanoTX-v3.5.0(external), RX:EPW6-v3.5.0 > OK
From these results, the cause appears to be in the internal TX module firmware.

chofchop, 2024-09-09:
> Just increase the UART speed for the internal module.

Are you talking about the baud rate?
As far as I understand, anything above 1.87M is fine.

chofchop, 2024-09-09:
I first suspected a poor contact in the wiring and collector of the internal module, but after seeing that the problem did not occur with the v3.3 firmware, I began to think that this was a software problem rather than a hardware problem.

pkendall64, 2024-09-09:
> That being said, I don't think it's a problem that can be ignored.

The ESP8285 is old and is nearing Espressif end-of-support. The ESP32C3 is the replacement going forward.
Using an RX as a TX is a convenience not that something that the devs actively support.

> Further test results.
> TX:EP1tcxo-v3.5.0, RX:EPW6-v3.5.0 > NG
> TX:EP1tcxo-v3.5.0, RX:EPW6-v3.3.2 > NG
> TX:EP1tcxo-v3.3.2, RX:EPW6-v3.5.0 > OK
> TX:BetaFPVnanoTX-v3.5.0(external), RX:EPW6-v3.5.0 > OK 
> From these results, the cause appears to be in the internal TX module firmware.

Basically all this says is that ESP8285 as a TX is not working that well with 3.5 onwards.

I will however continue to look at this issue.

chofchop, 2024-09-09:
> The ESP8285 is old and is nearing Espressif end-of-support. The ESP32C3 is the replacement going forward. Using an RX as a TX is a convenience not that something that the devs actively support.

I am very sorry for placing a burden on the developers.
Please let me know if there is anything I can do to help.

chofchop, 2024-09-12:
When I investigated, the problem did not occur up to ac355bf, but it seems that the problem has clearly occurred since 9cf3388.
I hope this is of some reference.

TX hardware: EP1tcxo, BetaFPV Nano RX
RX hardware: EPW6tcxo
Handset model: MT12
EdgeTX version (including nightly number) : 2.10.4
Packet Rate: F500
Telemetry Ratio: 1:16
PWM frequency: 333Hz,333Hz,160Hz,50Hz,50Hz,50Hz

chofchop, 2024-09-23:
I confirmed that the problem continues to occur even in v3.5 (not RC). This is just a report.

MUSTARDTIGERFPV, 2025-03-24:
Closing this issue for now since we've not heard anything for the past few months and can't reproduce this with supported hardware.

closed 2025-11-16 by github-actions[bot]