ExpressLRS/ExpressLRS #2923: [V3.5.0-RC2] Failsafe issue

0crap, 2024-09-09:
## Current Behavior

After installing 3.5.0-RC2 I had a failsafe happening

## Steps to Reproduce

Flash both TX and RX to 3.5.0-RC2 and go fly.

## Possible Solution (Not obligatory)

-

## Details

I did two flights with my Dolphin on RC2, first flight had a failsafe, second flight went normally.
https://youtu.be/xOGtjDmgGMQ
[image]

I also flashed my quad with RC2 which gave multiple "Telemetry lost" messages, but the link did not failsafe.
The quad uses a HM EP1. (Flashed with "betaflight passthrough" option.)
I was able to flash the quad (HM EP1) back to v3.4.3 which solved the issue.
Dolphin is flashed back to v3.4.3 as well but no time to fly it again atm.

The quad as well as the plane have had many glitch free flights on v3.4.3

## Your Environment

- TX hardware: DIY2400 E28 ISM2G4
- RX hardware: HM EP1 Dual TCXO
- Handset model: Taranis QX7S with TBS hardware mod (1.87MBps)
- OpenTX version (including nightly number): 2.3.15
- ExpressLRS version (TX & RX MUST MATCH): 3.5.0-RC2
- Packet Rate: 150Hz LoRa
- Telemetry Ratio: Std (1:32)
- user_defines:

0crap, 2024-09-09:
For the record, I was able to sqeeze in two flights with the Dolphin plane this morning, TX and RX on v3.4.3.
No issues.

0crap, 2024-09-10:
[image]

Here we have a part of the OpenTX logging on my handset at the moment of the failsafe.
(RTC of my handset was behind around 7 minutes, so forget the time stamps.)
Usually i receive one logline roughly every second, at the moment of the failsafe there is a gap of around 10 seconds.
So I guess the OTX logfile is of little help here...

pkendall64, 2024-09-10:
That sounds like a reboot of either the TX module or the RX.

0crap, 2024-09-10:
Is there any way I as a user could have noticed which one rebooted? (If that happened.)

pkendall64, 2024-09-10:
If you have OSD data as well you should be able to tie it up.
If the RX rebooted then you should see in the OSD that the LQ immediately goes to zero. If the TX rebooted then the LQ on the OSD would drop rapidly to zero but you will see the values as it drops.

0crap, 2024-09-10:
OK, well, I have nothing more besides the OSD data I used to render the YT video linked above.

0crap, 2024-09-11:
Yep, already done. :-)

JyeSmith, 2024-09-11:
Have you been able to bench test and reproduce?

0crap, 2024-09-12:
Sorry, not yet.
I'm away from my gear until after the weekend.
I'll report back next week.

0crap, 2024-09-19:
I tested this a bunch of times on the bench.
Connect/disconnect a RX with some time in between to see if it disconnects.
I was not able to reproduce it.

If other users read this and had a failsafe, please add your findings to this issue.
Otherwise I'm probably alone here and we can only close this issue and see how it goes....

JyeSmith, 2024-09-19:
We'll leave it open for a bit since RC2 will become the final release.  So we might suddenly get a whole lot of more users and some further info.

closed 2024-11-03 by github-actions[bot]