ExpressLRS/ExpressLRS #2935: Airport link doesn't recover breaks without power cycling both sides.

sevet, 2024-09-17:
## Current Behavior

Airport link doesn't recover breaks and link sometimes just breaks (sort of 2 issues), Only recovers on reset of both sides.

## Steps to Reproduce

Creat airportlink, make sure you have a link, power cycle one side and the link doesn't recover.
This was also mentioned by Pawel Spychalski  about 3 month ago here:
[](https://www.youtube.com/watch?v=f8jwl-f2RLA)

Also a different issue, it will be great if Airport will work with regular RC like the new Mavlink support, not needing 2 sets of TX/RX for Airport and control.

## Your Environment

- TX hardware: Radiomaster Bandit Micro, connected to TX/RX pads.
- RX hardware: iFlight diversity 500mw RX
- Handset model: No handset.
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

JyeSmith, 2024-09-19:
What FC firmware are you using?

sevet, 2024-09-19:
Hi,

No fc, just direct serial link to onboard computer.

sevet, 2025-01-09:
Airport is Perfect on 3.53, recovers even from very long link breakage.

closed 2025-01-09 by sevet