ExpressLRS/ExpressLRS #2938: Support CRSF_FRAMETYPE_SUBSET_RC_CHANNELS_PACKED

wimalopaan, 2024-09-19:
Support https://github.com/crsf-wg/crsf/wiki/CRSF_FRAMETYPE_SUBSET_RC_CHANNELS_PACKED packets.

This can be limited e.g. to max. 16 channels per packet.

mha1, 2024-10-22:
0x17 has a different scaling than 0x16 and doesn't support extended limits.

mha1, 2024-11-27:
Recently published TBS CRSF spec (https://github.com/tbs-fpv/tbs-crsf-spec/blob/main/crsf.md) discourages implementation of 0x17.

```
Ox17 Subset RC channels packet
Warning
This frame is discouraged for implementation. Revision is in progress.
```

I suggest to close this issue and wait for the revised 0x17 specification.

wimalopaan, 2024-12-03:
Do you really expect TBS to release a new version of the specs in the next e.g. 3 years?

I think there should be a way to transmit more than 16 channels over one link.

pkendall64, 2024-12-03:
You're right, there should be. But using a message type that is already discouraged is not the way to go.

wimalopaan, 2024-12-03:
So. what are the alternatives? 
As I understand https://github.com/crsf-wg should be the guideline.

1) use 0x17 as long as TBS does not publish a new version, which does not only deprecate the use, but defines a new meaning
2) create a new protocol extension 

Either way there are (minor) incompatibilities. Both of them could be circumvented if the handset (EdgeTx) does not use these. So, a user can enable that in EdgeTx and she should know that on the receiver-side this use must also be enabled.

closed 2025-01-11 by github-actions[bot]