ExpressLRS/ExpressLRS #2942: RadioMaster Nomad 2.4/900 TX - WiFi connectivity issues with Web UI

Martivip, 2024-09-20:
Hi after clean flash of the 3.5.0-RC2 or 3.4.3 to my new RadioMaster Nomad 2.4/900 TX
I am experiencing issues with WiFi connectivity of the module. There is no mater how I enable WiFi mode via LUA or when the module is outside the radio and it enters to WiFi mode by them self after 60 Seconds. When it enter to WiFi the network ExpressLRS is established, and I can connect to that network. But when I connect to web UI then sometimes is very fast and responsive as it should but most of the time web UI won't load at all or is slow and sluggish :(

I have other Beta FPV SuperG module and there everything works as it should, it connects quickly to Web UI and the UI is very responsive.

JyeSmith, 2024-09-20:
I just ran through flashing 3.4 and 3.5RC2 on a new module.  Both worked as expected.  For the record my home router is about 5m away. 

Can you also try Erase before flash?

Martivip, 2024-09-20:
Already did, no improvement ok then I will contact Radiomaster it looks like my unit is faulty one.
One more thing did you said that you connecting the module to your router and this is now that I do, in my case module establishing their network ExpressLRS that I connecting. Please try this scenario

JyeSmith, 2024-09-20:
This was the first WiFi test before flashing my home WiFi.  It worked no problems.

Martivip, 2024-09-20:
Ok thanks for checking, so my unit looks faulty. OK then I will contact Radiomaster to replace my unit.

zyren, 2024-09-21:
Actual tests have found that firmware versions after 3.4 are more likely to be interfered by routers when using WiFi. It will appear that the SSID cannot be searched, or the WEB page cannot be opened after connecting. If you stay away from the router or temporarily turn off the router, it can be used perfectly, or add the ELRS module. Home network, it is more convenient to use http://elrs_tx.local to access.

The metal casing of Nomad is also one of the reasons why the WiFi signal is relatively weak, so considering these factors, when connecting to the Nomad WiFi hotspot, you can also let the computer say that the phone is as close to the module as possible, which will get good results.

Martivip, 2024-09-21:
@zyren this is not applicable on my case, module is few centimetres behind my laptop. I also switched off my router and no improvement. As I'm said my all other ELRS modes works perfectly. As you see Jye from dev Team is not able to replicate the issue. I've asked also on discord and users also don't have issues and that leads to conclusion than my unit is faulty one unfortunately.

Martivip, 2024-09-22:
@JyeSmith this is really interesting when I remove the board from casing it connects via WiFi immediately even from different room, so definitely there is some hardware issue with my module. Maybe my unit has bad PCB antenna. I will report that to Radiomaster

closed 2024-09-23 by Martivip

S34nYun, 2024-09-26:
@Martivip I have the same symptoms as you. Has your Nomad product been replaced? When the WebUI doesn’t open after connecting to ExpressLRS WiFi, bringing my phone close to the Nomad device helps it open.

Martivip, 2024-09-26:
> @Martivip I have the same symptoms as you. Has your Nomad product been replaced? When the WebUI doesn’t open after connecting to ExpressLRS WiFi, bringing my phone close to the Nomad device helps it open.

Lucky you, my don't connects even is sitting on the phone or laptop, please contact their support they will help you

douglasnoronha, 2024-09-26:
I had the same problem, after a few attempts I managed to access it, I noticed that it was slow too.

Martivip, 2024-09-27:
> I had the same problem, after a few attempts I managed to access it, I noticed that it was slow too.

this is really interesting, there is manu of us that don't have any issues and so many that struggles with this WiFi connectivity  issues. Radiomaster said that after my report they verified some sample of production units to ensure this is one off issue, of course I don't know what are their conclusions in this matter but I believe that Radiomaster will always react professionally and will fix the issue. Please contact their support.

S34nYun, 2024-09-27:
@Martivip Thank you for your attention to my issue.
I have submitted support tickets to RM as follows. 
-> #20292 The WEBUI does not open or is very slow after connecting to the Nomad product's WIFI in AP mode.
May I ask how your issue was resolved?
I have other brands of ExpressLRS TX, but I have no issues opening the WEBUI like I do with the Nomad.
From the above post, there are mentions of issues like WIFI signal interference or antenna problems with the circuit board. 
I measured the signal strength of the Nomad product and another ExpressLRS TX product at the same distance in WIFI AP mode, and it doesn’t seem to be a WIFI signal issue.
[image]
There seems to be no problem connecting to WIFI with my phone or laptop in AP mode. 
The WEBUI sometimes opens well, but when refreshing the page or reconnecting to WIFI, it seems that the WEBUI either does not open or loads very slowly, resulting in the page not displaying correctly.
This issue I'm experiencing is quite frustrating...
I will wait for RM's response.

Martivip, 2024-09-27:
My unit faulu unit will be replaced. Signal strength is not only one factor that will provide stable connectivity. Please look the RRSI of both modules and compare.

Martivip, 2024-09-27:
My unit will be replaced. Definitely there are some faulty units in the our hands. But it could happen in a mass production unfortunately.
I know that Radiomaster will resolve your issue as well.

S34nYun, 2024-09-28:
After some time following the initial activation of AP Mode WiFi, I can no longer open the WEBUI at all. The WiFi connection remains stable, but packets between the ELRS TX are being dropped.

[image]

The distance between the nomad and the laptop is less than 1 meter, and there is no interference from any WiFi signals. (Other ELRS TX units work perfectly fine.) The WEBUI is not opening properly because packets are being dropped.

Whether it's an issue with the ESP32 chip... I am waiting for a response from RM.

douglasnoronha, 2024-09-28:
> After some time following the initial activation of AP Mode WiFi, I can no longer open the WEBUI at all. The WiFi connection remains stable, but packets between the ELRS TX are being dropped.
> 
> ![ping](https://private-user-images.githubusercontent.com/25032437/371773598-beaa4c7e-b018-4dde-b7eb-e0a5cb3c1d9f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc1NTk1ODYsIm5iZiI6MTcyNzU1OTI4NiwicGF0aCI6Ii8yNTAzMjQzNy8zNzE3NzM1OTgtYmVhYTRjN2UtYjAxOC00ZGRlLWI3ZWItZTBhNWNiM2MxZDlmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI4VDIxMzQ0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIzNmY4MThkZjczZTYwNTYzZGViZGZiYWY0NDQ5MWRhNjI4NzJhNjI4MjFjMmU0OTE2MjM5NmFmODFmNzM3MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.DE0jWr0yctS1dFP0sKpIpnmnClipfD7eryfmVU6MHxU)
> 
> The distance between the nomad and the laptop is less than 1 meter, and there is no interference from any WiFi signals. (Other ELRS TX units work perfectly fine.) The WEBUI is not opening properly because packets are being dropped.
> 
> Whether it's an issue with the ESP32 chip... I am waiting for a response from RM.

[image]
I have the same situation, nomad module on the side less than 1m away.
I will also open a ticket to check the situation.

S34nYun, 2024-09-29:
@douglasnoronha The symptoms of the issue are the same as mine. 
If packets are being dropped, it’s understandable that the web UI wouldn’t open properly.
It seems that further investigation is needed regarding this issue.
And, RM responded that the seller would provide direct assistance if contacted, and my ticket has been closed.
Right now, I have sent a message to the seller about my issue and am waiting for a response.
I will share updates on the progress of my issue. 
Thank you.

douglasnoronha, 2024-10-06:
Today I managed to solve the problem of accessing the webui in the nomad module.
I installed version 3.4.0 via serial and activated the option to Erase before flash.
I tested access via WiFi and quick access, ping working normally.
I then did the same with 3.4.1, 3.4.2, 3.4.3, all working perfectly, although even in these versions I had the same problem.
So I updated to 3.5.1 and it also worked perfectly.
Even the ping through the home network is better than before.
Do the update via serial and activate the option to Erase before flash.

Martivip, 2024-10-07:
> Today I managed to solve the problem of accessing the webui in the nomad module. I installed version 3.4.0 via serial and activated the option to Erase before flash. I tested access via WiFi and [...]

Unfortunately for me thaw was not the case, I've tried to flash with Erase before flash option and don't helped at all.

Martivip, 2024-10-09:
Today I received the replacement unit, and WiFi works like a charm.

douglasnoronha, 2024-10-10:
[image]
This is mine after the update and selected "Erase before flash"