ExpressLRS/ExpressLRS #2946: Radiomaster Bandit TX module isn't being detected by TBS Tango 2 since v3.5

Martin-Schlodinski, 2024-09-22:
## Current Behavior

Hi!

I just updated my Radiomaster Bandit TX from v3.4 to v3.5 and together with it the LUA script. Since then I can't use the LUA script anymore on the TBS Tango 2. The display only shows the error message defined in line [909](https://github.com/ExpressLRS/ExpressLRS/blob/master/src/lua/elrsV3.lua#L909) and below: "No ExpressLRS. Enable a CRSF Internal or External module ..."

The workaround I found was to comment out line [942](https://github.com/ExpressLRS/ExpressLRS/blob/master/src/lua/elrsV3.lua#L942) where it tries to detect the module.

I saw this `checkCrsfModule` was just introduced with v3.5 and it doesn't seem to work on my TBS Tango 2.

After commenting out this line the script works flawless!

I'm happy to help debugging but don't know how: how can I list all available modules with their type? The script searches for `mod.Type = 5` so perhaps the type differs on my remote?

## Steps to Reproduce

Install v3.5 on a Radiomaster Bandit TX (probably TX modules have the same issue).
Attach the module to the TBS Tango 2 remote and try to run the v3.5 LUA script.

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: Radiomaster Bandit TX
- Handset model: TBS Tango 2 with latest FreedomTX version (1.3.0, just updated)
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): v3.5.0

pkendall64, 2024-09-22:
From looking at the FreedomTX source, which is not updated in github, they don't have the .Type value in the module data table returned by `model.getModule()`.
So it may be that we have to test if the Type is nil and assume it's all good.
[image]
Could you change the Lua script line 899 as shown and let us know.

Martin-Schlodinski, 2024-09-23:
wow thanks for looking into it! The change you suggested worked flawless! Shall I create a pull request?

pkendall64, 2024-09-23:
> wow thanks for looking into it! The change you suggested worked flawless! Shall I create a pull request?

Sure, I'll be happy to review it for you.

pkendall64, 2024-09-23:
Can you make the PR against the 3.x.x-maintenance branch and we will get that into 3.5.1

Martin-Schlodinski, 2024-09-23:
sure! Here you go: https://github.com/ExpressLRS/ExpressLRS/pull/2950

closed 2024-09-24 by Martin-Schlodinski