ExpressLRS/ExpressLRS #2962: PPM Mode ARM on Ch5: Conflicts with Built-In Mixes in Older Systems (Futaba, Hitec)

WarhawkCZ, 2024-09-29:
## Current Behavior
ELRS v3.4 added the support for PPM input. This allows using the ELRS Tx modules with older radio systems (Futaba, Hitec, Graupner, etc.). These systems are still very capable, especially for traditional R/C planes. However, many in-build wing mixes, (e.g. butterfly) collide with channel 5 that is reserved for ARMing. Typically, it is not possible to re-map the mix to a different channel.

## Steps to Reproduce
Use any kind of an older radio as the signal source for ELRS Tx Module.

## Possible Solution (Not obligatory)
Change the order of channels such the Ch5 for ARMing is the last one. This one is typically free. 
 Now: CH1, CH2, CH3, CH4, CH5, CH6, CH7, CH8, CH9
  New: CH1, CH2, CH3, CH4, CH6, CH7, CH8, CH9, CH5
[Alternatively, clarify in the manual if it is safe to permanently operate the system ARMed and provide the solution to enforce so.
](https://github.com/ExpressLRS/ExpressLRS/discussions/2961)

## Details

Original PPM Pull request:
https://github.com/ExpressLRS/ExpressLRS/pull/2528

## Your Environment

- TX hardware: Futaba T9CAP with Jumper Aion ELRS 2.4G TX Nano
- RX hardware: any 
- ExpressLRS version (TX & RX MUST MATCH): v3.4 FCC

Many thanks to all of you for consideration and your hard work.

[image]
[image]
[image]

stanislavdavid, 2024-09-30:
It can by solved by PR #2801  or #2869

WarhawkCZ, 2024-09-30:
> It can by solved by PR #2801 or #2869

This is excellent reply @stanislavdavid . I actually like the Limits feature as well as the radio misses it.
It would be great adding these PRs.

OzDruiD, 2024-10-02:
> It would be great adding these PRs.

Yess! in particular, https://github.com/ExpressLRS/ExpressLRS/pull/2869 is a game changer for a lot of things.
A lot of older FBLS as well, such as original original BeastX can use Sbus, but not the SUMD implementation as just 1 of many this will fix.