ExpressLRS/ExpressLRS #2973: Mavlink telemetry works but does not retrieve parameters

maciek252, 2024-10-07:
## Current Behavior
I've managed to set up mavlink RC and telemetry. It works nicely, however the parameters are not retrieved. QGroundControl complains with a message "Vehicle 1 did not answer the request about parameters". The progress bar starts loading parameters but gets stuck at ~40%.

## Steps to Reproduce

1. Bind the link
2. Connect to the QGroundControl on a mobile phone via UDP port 14555
3. Watch as parameters start downloading but never end.

- TX hardware: TX: BetaFPV TX 2.4 GHz 1W 
- RX hardware: RX: BetaFPV Nano Rx 2.4 GHz
- Handset model: (Android 11, not relevant)
- OpenTX version (including nightly number): the last one, not relevant
- ExpressLRS version (TX & RX MUST MATCH): tested with versions 3.5.0 and 3.5.1, and backpack 1.5.1
- Packet Rate: 2:1
- Telemetry Ratio: 333 MHz? (full set of channels but the same was with 4 first channels full range PWM)
- user_defines:

MrMinecraft97, 2024-10-08:
Isn't the port 14550?

maciek252, 2024-10-08:
> Isn't the port 14550?

No, it does not work with 14550 but works with 14555. Not quite as written in the docs but I've tested this.

MrMinecraft97, 2024-10-08:
Make sure to update both RX and TX to 3.5.1. The .1 fixed a major bug in how mavlink uplink packets are sent to the plane from the gc software.

maciek252, 2024-10-08:
When I've read the release notes for 3.5.1 I hoped a lot that the problem will be gone. Unfortunately there was no improvement, I've tested both Rx and Tx running 3.5.1. Maybe it's related to Q_ENABLED=1? (VTOL options) I'll test this with Q_ENABLED=0.

sunfengxin0, 2024-10-27:
I also have the same problem

sunfengxin0, 2024-10-27:
but it work well in mission planner

menschel, 2024-11-21:
Just saw this Issue when I was about to file an issue on 900Mhz Mavlink with QGC.

I have a perfectly working setup with 2G4.
RM TX12MK2 internal ELRS + BetaFPV ELRS lite receiver both 2G4 ELRS 3.5.1 and Backpack 1.5.1.

Check your firewall, port 14550 must be open, port 14555 may be a high latency link which does not support mavftp and parameters.

On the other hand the 900Mhz system is completely broken for me.

maciek252, 2025-03-17:
Hello,

I hoped that the problem could be solved by new releases of ExpressLRS and QGroundControl, but there were hardly any and nothing has changed. I still get (relatively) reliable telemetry, but without ArduPilot parameters, also I cannot arm/disarm via QGroundControl etc

@menschel, could you please explain what does it mean that port 14550 must be opened? I've defined two UDP connections in QGroundControl, one over the port 14550 and another one 14555, and also sometimes QGroundControl connects via the default UDP connection (which is shown in the top right corner). However, the parameters do not work in any case. 

For some reasons (using the technology for military purposes?) the development of ExpressLRS and QGroundControl has stalled, which is quite surprising expecially for ELRS, where the new releases used to happen much more often...

menschel, 2025-03-17:
Set up an Access Point on your PC and have the ELRS TX connect to it by 
default. This is easily done via Configurator before flashing.

I assume you're on linux.

sudo ufw allow 14550/udp

will open up port 14550. You can limit it to a specific network device.

Delete your manually set connections in QGC, select the option to 
autoconnect on UDP and restart QGC.

Everything should work after that.

BTW I also share the impression that development is delayed in that 
context and this is likely Microsofts doing with their enforcing regime 
on Github.

BR

maciek252, 2025-03-18:
I've just checked this on linux and it worked out of the box, without the need to change any firewall rules. I've just connected to the expresslrs wifi network, launched QGroundControl and it loaded all the parameters, without even configuring the connection.

So android is the problem. And this is the most important part for me, as I wanted to be able to modify the parameters in the field. Recently, my trip was completely useless as I've set up some RC passtrough on a channel and ArduPilot did not like it and did not pass the safety checks. OK so I'll ask about the problem in the QGroundControl forum, thanks for your answer!

BTW connecting to QGroundControl via USB works... sometimes...

closed 2025-03-24 by MUSTARDTIGERFPV