ExpressLRS/ExpressLRS #2976: ELRS 3.5.1 RM Boxer and SPI RX connectivity issue

Gozzz, 2024-10-09:
## Current Behavior

SPI RX on whoop board drops connection every few seconds. No model mismatch notification, I use model match but that doesn't seem to matter if it is on or off.

## Steps to Reproduce

Flash RM Boxer to 3.5.1, bind to SPI RX whoop with bind phrase.

## Possible Solution (Not obligatory)

Flash RM Boxer back to 3.5.0

## Details

## Your Environment

- TX hardware: RM Boxer
- RX hardware: SPI RX CRAZYBEEF4SX1280
- Handset model: RM Boxer
- OpenTX version (including nightly number) EdgeTX 2.10.3
- ExpressLRS version (TX & RX MUST MATCH): 3.5.1 BF 4.5.1
- Packet Rate: 500Hz
- Telemetry Ratio: Std (1:128)
- user_defines:

SsgDan94, 2024-10-14:
Just posted my issue as well. Also having frequent connection drop with 3.5.1

Gozzz, 2024-10-15:
> Just posted my issue as well. Also having frequent connection drop with 3.5.1

So it might be wider than spi rx whoops? I haven’t tried it on my 5” builds yet.

Gozzz, 2024-10-21:
https://youtube.com/shorts/d3fe_obF-So?si=E1rIInNjXFKkZEZ7

Here is what I'm seeing.

mha1, 2024-10-22:
> https://youtube.com/shorts/d3fe_obF-So?si=E1rIInNjXFKkZEZ7
> 
> Here is what I'm seeing.

What is baud rate set to?

Gozzz, 2024-10-22:
> > https://youtube.com/shorts/d3fe_obF-So?si=E1rIInNjXFKkZEZ7
> > Here is what I'm seeing.
> 
> What is your baud rate set to?

Internal module is set to CRSF 5.25m

mha1, 2024-10-22:
> > > https://youtube.com/shorts/d3fe_obF-So?si=E1rIInNjXFKkZEZ7
> > > Here is what I'm seeing.
> > 
> > 
> > What is your baud rate set to?
> 
> Internal module is set to CRSF 5.25m

The higher the better? Not always. The probability of bit errors in the serial data exchanged between handset and module increases with baud rate. Bit errors will be caught by CRCs but a frame with bit errors is lost.

So in reality: fast enough is better. For 500Hz and lower 921k is sufficient (400k isn't), for F1000 you need 1.87M. Please set baud rate to 921k for your 500Hz packet rate and try to reproduce the error again.

Gozzz, 2024-10-24:
> The higher the better? Not always. The probability of bit errors in the serial data exchanged between handset and module increases with baud rate. Bit errors will be caught by CRCs but a frame with [...]

Understood, I can try it but I'm just reporting there is an issue. This is not a problem on 3.4.3.

I run the same radio for 5" racing so it is not ideal to have to change the radio system settings when going back and forth between models that run higher/lower packet rates. I'll test at 1.87m and post results.

porlock, 2024-11-24:
Same story for 5in1 HDzero board.. on 3.5.2 disconnect every 3-5 seconds.
https://youtu.be/tKh3LXnVEwQ

pkendall64, 2024-12-29:
Could you guys test PR #3079 and place feedback?

Gozzz, 2025-01-01:
> Could you guys test PR #3079 and place feedback?

Flashed my Boxer with #3079 using elrs configurator and the link is now solid.

bg-anza, 2025-01-09:
> > Could you guys test PR #3079 and place feedback?
> 
> Flashed my Boxer with #3079 using elrs configurator and the link is now solid.

Hi. I'm having the same issue with my RM boxer. How do you compile the firmware with the fix applied? Or can you please give a youtube tutorial link to compile the firmware? Thank you.

pkendall64, 2025-01-09:
> Hi. I'm having the same issue with my RM boxer. How do you compile the firmware with the fix applied? Or can you please give a youtube tutorial link to compile the firmware? Thank you.

You can use configurator, just select the "GIT PULL REQUEST" tab, enter 3079 in the "Git pull requests" box and select the PR, then flash as per usual.

bg-anza, 2025-01-12:
> You can use configurator, just select the "GIT PULL REQUEST" tab, enter 3079 in the "Git pull requests" box and select the PR, then flash as per usual.

Thanks for the tutorial. I managed to flash my RM boxer with the fix, but the problem still occurs. FYI my rx is a SPI rx on Newbeedrone RS FC. It came with BF 4.5.0 KAACK. Trying to re-flash the firmware also did not help. 
Flashed it to BF 4.4.0 fix the issue and the connection is solid. But somehow it didn't work with BF 4.5.0 🤷‍♂️

Gozzz, 2025-01-12:
> Thanks for the tutorial. I managed to flash my RM boxer with the fix, but the problem still occurs. FYI my rx is a SPI rx on Newbeedrone RS FC. It came with BF 4.5.0 KAACK. Trying to re-flash the [...]

Is NBD still providing their own BF forks? Did you try reflashing BF 4.5? I ran 30+ packs last night without any issues on 2 different HM/Hdzero aio5s. Both running BF 4.5.1 with other options set to digital osd, vtx, and race pro during flashing on bf configurator.

pkendall64, 2025-03-12:
> There is a regression for this issue. When using BF 4.5.1 (HDZero AIO5) and Boxer with eElrs 3.5.3, it disconnects every few seconds while disarmed; once armed, it's okay, until disarm **There is no issue with Elrs 3.4.3**

This issue is already fix and will be available in the next release. You can test it by flashing the 3.x.x-maintenance branch.

kirek007, 2025-03-12:
I thought it was already released. Then I'll test code from the branch. Thanks

JSMiCh, 2025-05-25:
> This issue is already fix and will be available in the next release. You can test it by flashing the 3.x.x-maintenance branch.

I have the same issue with HDz AIO and Boxer, flashed the 3.x.x-maintenance branch, it didn't work. Rebinding only works (

closed 2025-05-25 by pkendall64

JSMiCh, 2025-05-26:
Yes, I flashed 3.5.5 to TX, 4.5.2 BF, last EdgeTX, last ELRS backpack and problem still exist. But at least it goes away when you manage to arm a drone

JSMiCh, 2025-05-26:
> This fix was released in 3.5.4, you need to reflash the transmitter module to at least that version.
Just flashed to 3.5.4 and it worked, but the bug appears on 3.5.5 :\

JSMiCh, 2025-06-06:
> > > This fix was released in 3.5.4, you need to reflash the transmitter module to at least that version.
> > 
> > 
> > Just flashed to 3.5.4 and it worked, but the bug appears on 3.5.5 :\ again
> 
> I can't replicate this, I am on 3.5.5 on my Boxer internal module and my link is solid on my AIO5 whoops.

Ok, i just flashed again to 3.5.5. Bug is back (at least until successful arm). Maybe I can provide some additional info?

PR #3079 (2025-01-23): Fix SPI FC problem with looping connection loss, reestablishment
This is a fix for the issue with Betaflight ELRS SPI flight controllers that exhibit an issue when using the VTX admin.
Upon RF connection, the TX sends the current VTX settings and a EEPROM save, which causes the link to drop. The connection is reestablished, and we send another VTX packet with EEPROM save ad infinitum.

The fix is to properly detect disconnected state and start the debounce timer so if the connection is reestablished before the debounce time (1 second) the VTX Admin packets are not sent again, thus stopping ano [... cut]