ExpressLRS/ExpressLRS #2982: ELRS 3.5.1 - Airport  Mavlink does not connect

NikosSprocket, 2024-10-12:
## Current Behavior

I have 2 Quads fully operational with Ardupilot 4.5.5 and 4.5.6 using the following equipment and ELRS 3.4.3 for the Airport Tx and Rx.
I tried multiple times flashing both Tx and Rx with ELRS 3.5.1 and Mavlink failed every time.
Re-flashed both Tx and Rx to ELRS 3.4.3 and Mavlink is working again.

Radiomaster Ranger 2,4GHz
Radiomaster  RP4TD True Diversity 2.4GH
F1000Hz 1:2
UART baud 19200
Ardupilot 4.5.5 and 4.5.6
Mission Planner 1.3.82
Silicon Labs CP210x USB to UART bridge

JyeSmith, 2024-10-16:
Confirmed and we are looking into it.

If you are using Mission Planner and Mavlink Id strongly recommend use the new mavlink protocol ion 3.5.  It has a number of advantages over airport.

JyeSmith, 2024-10-17:
@NikosSprocket can you please test this PR.

https://github.com/ExpressLRS/ExpressLRS/pull/2988

NikosSprocket, 2024-10-17:
Of course; sorry for the stupid question. Is this patch available on the mainstream 3.5.1 firmware or on nightly. If it is the nightly can you let me know where can I get it from?

JyeSmith, 2024-10-17:
> Of course; sorry for the stupid question. Is this patch available on the mainstream 3.5.1 firmware or on nightly. If it is the nightly can you let me know where can I get it from?

In the Configurator there is a tab for flashing Pull Requests.  Select the PR I linked above and flash both the tx and rx as normal.

> Also, would be possible to elaboratre why the single Tx/Rx Mavlink is better than separate RC and Mavlink comms?

The Mavlink protocol has packet acknowledgment and retries.  So If you drop a packet over the air the rx can request it be resent.  Unlike Airport where there are no retries.  This greatly improves the packet Quality in mission planner as LQ drops.

There is also no loss in performance sending RC over the same link.  The Data rate in the uplink from MP to the craft is very low, unlike the downlink which contains a lot of data.  This leaves lots of spare packets for RC.

NikosSprocket, 2024-10-18:
Good news! the "Fix Airport communication #2988" is working and was able to connect using the above equipment and software versions, besides ELRS ver. of course.

One thing I observed during an accidental test (user error!) is that the RadioMaster Ranger 2.4GHz rebooted every time I tried to "Connect" the MP in the following situation. The Ranger is setup for Airport UART 19200 but the  RP4TD is **_not_** in Airport mode. After configuring the RP4TD for Airport mode and 19200 UART it worked fine. I repeated the accidental test using ver. 3.4.3 
and got exactly the same results.

Thank you for fixing it and final question; so Airport because is using TLM will never be able to include some error correction or so. The only reason Airport was a good solution for UAS that it is able to have a separate communications hardware for MP this way the UAS can be still controlled via Mavlink just in case RC failed.
[image]
[image]

JyeSmith, 2024-10-18:
> Good news! the "Fix Airport communication #2988" is working and was able to connect using the above equipment and software versions, besides ELRS ver. of course.

💪 

> One thing I observed during an accidental test (user error!) is that the RadioMaster Ranger 2.4GHz rebooted every time I tried to "Connect" the MP in the following situation. The Ranger is setup for [...]

This is probably because the connection attempt is timing out and you have the below option in CONFIG>PLANNER checked.
[image]

> Thank you for fixing it and final question; so Airport because is using TLM will never be able to include some error correction or so. The only reason Airport was a good solution for UAS that it is [...]

You should still be able to use the mavlink protocol on its own with the separate communications hardware.

closed 2024-10-18 by JyeSmith

nithinteja1, 2024-10-19:
Hi jye. Thank you for the patch . I’m using radiomaster bandit micro 915mhz with br3 receiver. I tried this pr with 4800 baurd rate  works well and getting heart beats in mission planner . But when I change the baurd rate to 9600 only receiving single heart beat and getting timed out . I kept the packet rate at 200hz with 1W power output . I tried different Baurd rates nothing working. Can you please update on this….   And after flashing this PR the RX webui for config seems to have bug , it took 10-12 power cycles to get it work and had to wait 2hours to load the config page.

pkendall64, 2024-10-19:
https://www.expresslrs.org/software/airport/?h=airport#data-rate-vs-baud-rate

As can be seen in the docs the max baud-rate for 200Hz packet rate is 4800 baud.

nithinteja1, 2024-10-19:
Hi yes I sAW that table before . Is there other ways around? Because 4800 is not optimal , the latency in the mission planner is high even with simple setup , I’m trying to get telemetry for longer range models . I can’t use 2.4ghz elrs.

JyeSmith, 2024-10-19:
Use new lr1121 hardware (Radiomaster hardware and the DB rx) for faster subGHz modes.

NikosSprocket, 2024-10-19:
I do not see the PR "Fix Airport communication #2988" in the Configurator anymore.

JyeSmith, 2024-10-19:
It has been merged and is available in the 3.x maintenance branch.  This will become 3.5.2 soon.