ExpressLRS/ExpressLRS #2989: Using the `hidden` flag on a CRSF parameter crashes the `elrsV3.lua`

wimalopaan, 2024-10-17:
Adding the `hidden` flag (`0x80`) to a parameter according to https://github.com/crsf-wg/crsf/wiki/CRSF_FRAMETYPE_PARAMETER_SETTINGS_ENTRY actually crashed the `elrsV3.lua` script.

pkendall64, 2024-10-17:
We use that flag quite a lot, so I don't see how it crashes the Lua script.
If you go into the backpack menu and turn off the backpack all the entries become hidden (by the TX setting the hidden flag). Also, the Bad/Good is a hidden parameter that is sent all the time to the Lua script.

wimalopaan, 2024-10-17:
> We use that flag quite a lot, so I don't see how it crashes the Lua script.

I apologize, the error was on my side!

Now taking a LA trace and looking at the transferred values make the error clear.

closed 2024-10-17 by wimalopaan