ExpressLRS/ExpressLRS #2990: ELRS Rx and Tx drop local WiFi within a few seconds and transmit in AP mode

NikosSprocket, 2024-10-18:
## Current Behavior

For a while now all my RadioMaster Rx and Tx refuse to stay connected to the local (home) WiFi for more than a few seconds and then turn on AP mode which it works 100%. After connecting to the Rx/Tx AP network and examine the http://elrs_[tx,rx].local/ (10.0.0.1) WiFi tab the local WiFi SSID is displayed; see attached screenshots.
Finally, the Radiomaster Ranger 2.4GHz has an additional error about loosing it's auth. after the initial connection.

I have captured the routers Logs during the time the Rx or Tx WiFi turns on for a couple of receivers below:

RadioMaster RP4TD True Diversity 2.4GHz RX Firmware Rev. 3.4.3:
Oct 17 23:06:22 wlceventd: wlceventd_proc_event(530): eth2: Auth 14:2B:2F:AA:8B:74, status: Successful (0), rssi:0
Oct 17 23:06:22 wlceventd: wlceventd_proc_event(559): eth2: Assoc 14:2B:2F:AA:8B:74, status: Successful (0), rssi:-38
Oct 17 23:06:31 wlceventd: wlceventd_proc_event(511): eth2: Disassoc 14:2B:2F:AA:8B:74, status: 0, reason: Disassociated due to inactivity (4), rssi:0
Oct 17 23:06:31 wlceventd: wlceventd_proc_event(511): eth2: Disassoc 14:2B:2F:AA:8B:74, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
**Power-cycled and tried again:**
Oct 17 23:15:55 wlceventd: wlceventd_proc_event(530): eth2: Auth 14:2B:2F:AA:8B:74, status: Successful (0), rssi:0
Oct 17 23:15:55 wlceventd: wlceventd_proc_event(559): eth2: Assoc 14:2B:2F:AA:8B:74, status: Successful (0), rssi:-60
Oct 17 23:16:04 wlceventd: wlceventd_proc_event(511): eth2: Disassoc 14:2B:2F:AA:8B:74, status: 0, reason: Disassociated due to inactivity (4), rssi:0
Oct 17 23:16:04 wlceventd: wlceventd_proc_event(511): eth2: Disassoc 14:2B:2F:AA:8B:74, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0

RadioMaster Ranger 2.4GHz TX Firmware Rev. 3.4.3
Oct 17 23:18:14 wlceventd: wlceventd_proc_event(530): eth2: Auth B0:B2:1C:4E:5C:84, status: Successful (0), rssi:0
Oct 17 23:18:14 wlceventd: wlceventd_proc_event(559): eth2: Assoc B0:B2:1C:4E:5C:84, status: Successful (0), rssi:-58
Oct 17 23:18:37 wlceventd: wlceventd_proc_event(494): eth2: Deauth_ind B0:B2:1C:4E:5C:84, status: 0, reason: Disassociated due to inactivity (4), rssi:-65
Oct 17 23:18:37 wlceventd: wlceventd_proc_event(494): eth2: Deauth_ind B0:B2:1C:4E:5C:84, status: 0, reason: Previous authentication no longer valid (2), rssi:-65
Oct 17 23:18:38 wlceventd: wlceventd_proc_event(511): eth2: Disassoc B0:B2:1C:4E:5C:84, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0

Flywoo EL24P 2.4GHz RX Firmware Rev. 3.4.1 with similar Logs:
Oct 17 22:47:29 wlceventd: wlceventd_proc_event(530): eth2: Auth 8C:CE:4E:47:C6:4F, status: Successful (0), rssi:0
Oct 17 22:47:29 wlceventd: wlceventd_proc_event(559): eth2: Assoc 8C:CE:4E:47:C6:4F, status: Successful (0), rssi:-75
Oct 17 22:47:39 wlceventd: wlceventd_proc_event(511): eth2: Disassoc 8C:CE:4E:47:C6:4F, status: 0, reason: Disassociated due to inactivity (4), rssi:0
Oct 17 22:47:39 wlceventd: wlceventd_proc_event(511): eth2: Disassoc 8C:CE:4E:47:C6:4F, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Oct 17 22:47:43 wlceventd: wlceventd_proc_event(530): eth2: Auth 8C:CE:4E:47:C6:4F, status: Successful (0), rssi:0
Oct 17 22:47:43 wlceventd: wlceventd_proc_event(559): eth2: Assoc 8C:CE:4E:47:C6:4F, status: Successful (0), rssi:-75
Oct 17 22:47:52 wlceventd: wlceventd_proc_event(511): eth2: Disassoc 8C:CE:4E:47:C6:4F, status: 0, reason: Disassociated due to inactivity (4), rssi:0
Oct 17 22:47:52 wlceventd: wlceventd_proc_event(511): eth2: Disassoc 8C:CE:4E:47:C6:4F, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Oct 17 22:47:55 wlceventd: wlceventd_proc_event(530): eth2: Auth 8C:CE:4E:47:C6:4F, status: Successful (0), rssi:0
Oct 17 22:47:55 wlceventd: wlceventd_proc_event(559): eth2: Assoc 8C:CE:4E:47:C6:4F, status: Successful (0), rssi:-79
Oct 17 22:48:10 wlceventd: wlceventd_proc_event(494): eth2: Deauth_ind 8C:CE:4E:47:C6:4F, status: 0, reason: Disassociated due to inactivity (4), rssi:-78

[image]
[image]
[image]

pkendall64, 2024-10-18:
That is normal behaviour if the ESP cannot maintain a good connection to the router. i.e. the RSSI signal strength is too low so it disconnects and starts an access point.

NikosSprocket, 2024-10-18:
I placed the Flywoo and RP4TD 30 cm from the main router's antennas and the Ranger was 130 cm from the same antennas. If it was not the WiFi client then has to be something with the local WiFi APs. After looking further in the AP logs and scrutinized all the AP settings I found out that ASUS have added "Roaming Assistant" to their RT routers/APs. After turning off Roaming for the 2.4GHz network (5GHz still has Roaming enabled at -70 dBm) there was no problem with any of the previous Tx/Rx devices from staying connected to the local WiFi network.
Sorry for taking your time; I should have done more thorough troubleshooting before opening the case.

closed 2024-10-18 by NikosSprocket