ExpressLRS/ExpressLRS #2992: RCVR_UART_BAUD no effect in 3.5.0 and 3.5.1

Duese123, 2024-10-18:
## Current Behavior

RCVR_UART_BAUD is not overtaken the baudrate on UART

## Steps to Reproduce

Flash for FrSky R9 MM Mini newest SW 3.5.0 or 3.5.1 and change RCVR_UART_BAUD to 115200. Express Konfigurator 1.7.6 used

## Possible Solution (Not obligatory)

## Details

Receiver is still sending in default baudrate 420000 (checked with logic analyzer). Changes have no effect. Flash to 3.4.1 eg TX is sending correctly with baud 115200 flash back to newer version UART is again set to 420000

## Your Environment

- TX hardware: FrSky R9 MM Mini
- RX hardware: not relevant
- Handset model:not relevant
- OpenTX version (including nightly number) not relevant
- ExpressLRS version (TX & RX MUST MATCH): not relevant
- Packet Rate: not relevant
- Telemetry Ratio: not relevant
- user_defines: not relevant

JyeSmith, 2024-10-20:
@Duese123 can you please test this PR https://github.com/ExpressLRS/ExpressLRS/pull/2997

Duese123, 2024-10-21:
I was trying with new Branch "3.x.x-maintenance"
With Konfigurator but still not working

PR #2997 (2024-10-20): Fix STM32 flashing options
Fixes #2992
closed 2024-10-21 by Duese123