ExpressLRS/ExpressLRS #3006: Local build ignores build flag in user_defines

slackista, 2024-10-23:
## Current Behavior

Local build ignores build flag `-DRegulatory_Domain_EU_868`

## Steps to Reproduce

1. Select Git branch tab
2. Select master branch
3. Target: radiomaster.tx_dual.nomad.uart
4. Manual mode:
```
-DMY_BINDING_PHRASE="secret"
-DRegulatory_Domain_EU_868
-DUART_INVERTED
```
5. Logs don't print build_flags
6. Build and flash finishes without errors
7. TX is flashed successfully with AU915 reg domain

## Details
Configurator v1.7.6
Reproducible on ArchLinux and Debian11

pkendall64, 2024-10-24:
Can you attach the log from the build please?

slackista, 2024-10-24:
Build logs. Vanilla Ubuntu24.04.1
[expressslrs-configurator.log](https://github.com/user-attachments/files/17510932/expressslrs-configurator.log)