ExpressLRS/ExpressLRS #3009: ELRS receiver doenst respond / doesnt pass through device enumeration requests on CSRF protocol.

KissUltra, 2024-10-25:
## Current Behavior

When i try to enumerate devices on crossfire network, i get no responses from ERLS endpoint.
Most likely it doesnt respond or filter packets for some reason.

Broadcast messages must be propagated to ALL nodes of crossfire network. Over direct connection and over RF link.

## Steps to Reproduce

Have FC send enumeration broadcast message to 0 address.

uint8_t tmp[6] = {0xc8, 0x04, 0x28, address, CROSSFIRE_OSD_ID, 0}

CROSSFIRE_OSD_ID in my case 0x80, but u can try 0xc8 as well.

## Possible Solution (Not obligatory)

Crossfire broadcast packages must be properly routed through whole crossfire network.

## Details

I know, no osd/fc do that. But i do. Works like a charm with OG crossfire.

https://youtu.be/DugHg84Qv3M

## Your Environment

Kiss Ultra V2, Any ELRS receiver (latest fw)

- TX hardware: irrelevant
- RX hardware: any
- Handset model: irrelevant
- OpenTX version (including nightly number) irrelevant
- ExpressLRS version (TX & RX MUST MATCH): 3.5.1
- Packet Rate: 250
- Telemetry Ratio: 128
- user_defines: hah?

pkendall64, 2024-10-25:
I would assume that if the broadcast was implemented to work from the FC to the RX you would also be able to directly query the parameters from the RX. Also I assume the RX is supposed to forward the broadcast to the TX module and the FC could then also query all the parameters from the TX module. Should the TX then forward that to the handset? And the FC then be able to query all the handset parameters?

This would be a quite a large undertaking!

KissUltra, 2024-10-26:
Your assumptions are correct. Headset (radio) doesnt have config parameters and doesnt respond to  broadcasts. But! If you have USB connected crossfire device to crossfire TRANSMITTER it will be visible on the network. I even have Voltara ESC visible on crossfire (yes, all 4). Point is. Receiver must be a FULL "crossfire router" device. I know its not easy to make, but not hard either. I have router implemented on Ultra FC according to the specs.

KissUltra, 2024-10-26:
crossfire router MUST pass received broadcast packets to ALL endpoints connected to it. So for reciever ep1 will be FC. ep2 will be radio link.

KissUltra, 2024-10-26:
Also, router MUST maitain routing table that links device addresses and endpoints they are connected to. That is build up on any passing packet or on heartbeat packets. Router MUST deliver NON BRODCAST message to endpoint where device address registered on. Its fun! :)

pkendall64, 2024-10-26:
If only there was a public specification that we could all use to implement this "MUST" behaviour!

KissUltra, 2024-10-27:
I think you know now more then enough to do so :) i think you guys already kinda doing it partially, i can see devices from radio via crossfire agent lua. Just needs a bit more ;)

KissUltra, 2025-03-28:
Public specification is there for a few month now, could you guys allow enumeration of crossfire network from FC side?

KissUltra, 2025-03-28:
https://github.com/tbs-fpv/tbs-crsf-spec

pkendall64, 2025-03-28:
It's on my list of things to do. My list is very long though.

KissUltra, 2025-03-29:
Ok

KissUltra, 2025-03-29:
Let me know when its done, i can test it and enable crossfire tool for ELRS then. Thank you.

closed 2025-09-05 by pkendall64