ExpressLRS/ExpressLRS #3010: On PLATFORM_ESP8266 (at least), Serial.begin inverts both RX and TX polarity, rather than just inverting TX alone.

theglasshammer, 2024-10-28:
## Current Behavior

https://github.com/ExpressLRS/ExpressLRS/blob/master/src/src/rx_main.cpp:1368
Serial.begin(serialBaud, serialConfig, mode, -1, invert);

On the PLATFORM_ESP8266 (at least), Serial.begin inverts both RX and TX polarity, rather than just inverting TX alone, and therefore inverted CRSF doesn't provide telemetry.

## Steps to Reproduce

Configure Inverted CRSF on a receiver, observe missing telemetry from FC
 
## Possible Solution (Not obligatory)

USC0(UART0) |= BIT(UCTXI);

## Your Environment

Radiomaster RP1 V2 ExpressLRS 2.4ghz Nano Receiver

pkendall64, 2024-10-29:
When I added the CRSF inverted selection in the LUA I did test this solution and it didn't work for me at all!

theglasshammer, 2024-10-29:
Did you connect it to the SBus pin?

pkendall64, 2024-10-29:
Yes, of course. This was some time ago when I tried though.

closed 2025-01-12 by github-actions[bot]