ExpressLRS/ExpressLRS #3022: BLE Joystick not mapped correctly on Mac OS

fzlrhmn, 2024-11-06:
## Current Behavior
I was trying to connect BLE Joystick to Mac OS via bluetooth and found that the channel 3 and 4 not mapped correctly. It was recognized as channel 5 and 6. 

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2024-11-06:
Works fine for me on liftoff, tryp and DRL sim on the mac.
Does the simulator allow mapping the channels like the others do?

fzlrhmn, 2024-11-07:
> Works fine for me on liftoff, tryp and DRL sim on the mac. Does the simulator allow mapping the channels like the others do?

It doesn't have channel mapping unfortunately. I will try to reassign from radio later.

closed 2025-01-12 by github-actions[bot]