ExpressLRS/ExpressLRS #3023: Flashing 3.5.1 to RM RP2 bricks the RX

jabacz, 2024-11-07:
## Current Behavior

Flashing 3.5.1 to Radiomaster 2.4 RP2 RX bricks the receiver.
1. Flashed via Wifi -> Brick
2. Re-Flashed from bootmode via FTDI -> still brick
3. Re-Flashed 3.4.3 from bootmode via FTDI -> ubrick
4. try another spare RP2 from drawer not built in any model -> same results

## Steps to Reproduce

1. Flash 3.5.1 to RM RP2 by any method, you get bricked RX 

## Possible Solution (Not obligatory)

Recover using 3.4.3; FTDI; bootmode -> unbrick

## Your Environment

- RX hardware: Radiomaster RP2
- WiFi: RX with binding to home network saved
- FTDI:  generic chinese UART FTDI
- ELRS configurator: version issued with 3.5.1

pkendall64, 2024-11-07:
Could you try flashing with the FTDI and enable erase before flashing option.

jabacz, 2024-11-08:
@pkendall64 
Thanks
after full erase it works in all combinations
3.4.3 -> FTDI 3.5.1 -> WiFi 3.4.3 -> Wifi 3.5.1 -> WiFi 3.5.2
Some residual error in the flash from older versions?
I admit, these were upgraded several times using WiFi, no purge between

pkendall64, 2024-11-09:
Thanks for getting back to us. This could point to an issue in the config upgrader. I'll take a look.

closed 2024-11-09 by jabacz

jabacz, 2024-11-09:
Flash memory content when problem occurred
- Binding phrase set
- Home WiFi SSID + Password set
- Lock on connect ON
- ISM regulatory domain
Devices updated to new versions randomly several times a year since "prehistory" via WiFi