ExpressLRS/ExpressLRS #3030: SBUS trainer lags after a while (SBUS-inverted signal from ER-6 onto TX16S)

dirteat, 2024-11-11:
Hello there,

I have set up a trainer mode to connect 2 TX16S emitters. For this I am using:

- An ER-6 receiver, configured with SBUS inverted serial, 100Hz, and plugged onto the TX16S_Teacher with the TX port of the    ER-6 connected to the RX port of the TX16S_Teacher on AUX1
-  AUX1 on the TX16S_Teacher is set as "SBUS Trainer" and powered.
- The TX16S_Learner binds to the ER-6 receiver with password

Switching trainer mode on the TX16S_Teacher, I am receiving the input of the TX16S_Learner, first without issues (Chan or Stick). But after a while, tens of seconds, maybe a minute or two (a random time) the input start to lag, crazily lags (would be catastrophic while flying).

Video with lag in progress there:

[sbus_lagtest.webm](https://curl.irmp.ucl.ac.be/~chris/upload/aero/sbus_lagtest.webm)

## Workaround
Seems to not happen at 333Hz, seems to be loosely related to a loss of signal (telemetry loss pops from time to time when all these guys are close together).

## Details
I've opened that bug on EdgeTX first (I am a rookie to both edgetx and elrs), but people there seems to suggest that might be ELRS related:

https://github.com/EdgeTX/edgetx/issues/5646

Please have a look to that bug too, more details and additional tests are also given there!

## Your Environment

- TX hardwares: TX16S Mark II
- RX hardware: ER6
- Handset model: NA
- OpenTX version (including nightly number): EdgeTX 2.10.5 (all emitters all receivers)
- ExpressLRS version (TX & RX MUST MATCH): 3.5.1 (all emitters all receivers)
- Packet Rate: 100Hz full (333Hz seems stable)
- Telemetry Ratio: Std

Happy to test anything, as far as my skills go!
Best,
Chris.

JyeSmith, 2024-11-13:
Def not normal.  If this was a common issue we would be having a lot of feedback from sbus users.

To me it looks like a bad link (out of sync) between the TX16S_Learner and ER-6.  To confirm can you please record the test again with LQ shown on the TX16S_Learner (left Tx16 in your video).

If you can capture the change from normal to bad link that would be great.  Id like to see if the LQ changes, or TLM to the TX16S_Learner drops completely.

dirteat, 2024-11-13:
Noted, I'll try to do that and record it. Thanks!

mha1, 2024-11-13:
I could reproduce this on:

Slave radio: T20 3.5.2 ISM, BetaFPV Nano RX 3.5.2 100Hz, TLM std, 25mW
Master radio: TX16s. EdgeTX 2.10.5, internal ELRS 3.5.2 ISM

It doesn't need a slave rx. If it happens it's very obvious just looking at the channel monitor. It happens more or less random, I couldn't find a way to reproduce it systematically. If it happens it's on for quite a few seconds, sometimes more than 30s. LQ is perfect while it is happening.

I grabbed my LA to record the SBUS stream from the slave rx to the master radio while the channel monitor showed lagging stick movements. I converted the recorded SBUS stream to channel values to see if ELRS is the culprit. In the recording throttle was moved slowly from low to high with the TX16s channel monitor showing lagging movements. I can't support the theory ELRS is serving lagging or jerky SBUS data. Looks good to me. 

[image]

3djc, 2024-11-13:
So I guess you are not experiencing the signal loss issues that the OP is mentioning that made me think initially it could be more ELRS side. What freq are you running the internal module at ? (Since it drives the mixer runs)

mha1, 2024-11-13:
100Hz, 1:8, 8ch, model match off

mha1, 2024-11-13:
> It looks like there might be issues between mixer and sbus input timings

Yes, and the fact 333Hz works might be a hint

3djc, 2024-11-14:
I guess it can be closed here, since it looks now much more likely to be Etx, since you cleared some of the weird elrs related surrounding of the initial report

closed 2024-11-14 by JyeSmith