ExpressLRS/ExpressLRS #3034: When binding at 3.5.2, it will automatically become 50Hz and the binding cannot be completed at one time.

zyren, 2024-11-14:
## Current Behavior

## Steps to Reproduce

1. Open ExpressLRS lua Scripts on the remote controller.
2. Choose Packet rate randomly.
3. Press [Bind].
4. The rate in the connection status on the upper right corner shows 50.
5. The receiver has no response.
6. Change the Packet rate and it returns to the normal state. 

## Possible Solution (Not obligatory)

## Details

All the firmware was updated once using ExpressLRS Configurator, with Erase before checked.
All the firmware was also reflashed using Web Flasher, with Full chip erase checked.
External modules all used the Uart update method. Internal modules all used the EdgeTX Passthrough update method.
All the test results were the same.
Everything was normal on the 3.5.1 firmware.

## Your Environment

- TX hardware: Nomad, Boxer Internal Module, Bandit Nano
- RX hardware: DBR4-TD, ER6
- Handset model: TX16S MarkII, Boxer, Pocket
- EdgeTX version: 2.10.5
- ExpressLRS version (TX & RX MUST MATCH): 3.5.2
- Packet Rate: Any
- Telemetry Ratio: Any
- user_defines: Using ExpressLRS Configurator 1.7.6 and Web Flasher, no custom user_defines were used.

JyeSmith, 2024-11-14:
Can you please test this fixes the issue you see. https://github.com/ExpressLRS/ExpressLRS/pull/3035

zyren, 2024-11-14:
OK. I'll test it right away.

zyren, 2024-11-14:
> Can you please test this fixes the issue you see. #3035

Thank you very much. I compiled the firmware using #3035. Now everything is perfect. I tested Nomad, Boxer, Bandit Nano, DBR4, RP4TD, ER6, BR3, and everything is normal.

closed 2024-11-14 by JyeSmith