ExpressLRS/ExpressLRS #3036: Jumper T15 connection to exrpesslrs sometimes takes long

druckgott, 2024-11-15:
## Current Behavior

I have a Jumper T15 with the current Version of Edgetx 2.10.5.
And I have the current Version from ELRS 2.5.1.

I have sometimes problem that the connection between the T15 and the Expresslrs takes very long. I switch on the t15 and then switch on the drone.
Then there happens no connection. Only If I go to sys button and klick ExrepssLRS Button on the Menü then the LUa loads and the the connection starts. It looks like that the modul will not realy be aktivated/startup or something like this.

## Steps to Reproduce

    Start Jumper T15 with internal elrs modul.
    Start Drone
    Nothing happens
    Go to Sys Button and ELRS Menü --> connection and telemetry connects

## Possible Solution (Not obligatory)

I don´t know if its a problem on elrs or on edgetx:
https://github.com/EdgeTX/edgetx/issues/5667

## Details

## Your Environment

- TX hardware: Jumper T15 internal
- RX hardware: different Radiomaster RP Recivers
- Handset model: Jumper T15
- ExpressLRS version (TX & RX MUST MATCH): see above
- Packet Rate: different rates
- Telemetry Ratio: different ratios
- user_defines: no one

druckgott, 2024-11-16:
Its remove aß i understand
What happened to the Lua Init Rate option?[¶](https://www.expresslrs.org/info/init-rate/#what-happened-to-the-lua-init-rate-option)
The Init Rate receiver Lua option was removed in ExpressLRS 3.4 in favor of automatic Init rate storage.

mha1, 2024-11-16:
Yes, but why don't you read further (I prefer method 2):

[image]

druckgott, 2024-11-16:
perfekt thx this helps, but maybe it shoud be written on the top that it also needs to do on >3.5 ;-) otherwise it´s posible to stop reading

druckgott, 2024-11-16:
ok I did this but now both where off some time and the problem happens again

mha1, 2024-11-16:
Just out of curiosity:
- what's your baud rate set to (radio settings/hardware)?
- have you you changed the WiFi delay (default is 60 seconds)

What happens if you power the radio first, wait 10 seconds (important), then power the receiver, wait for the connect, don't do anything else. Does it connect eventually within 30-60 seconds or does not connect at all and the receiver goes to WiFi mode after 60s.

druckgott, 2024-11-16:
I teste Different from. 400 to 5.xx (dont know the exact number)
I did Not changed any thing on WiFi config

mha1, 2024-11-16:
> What happens if you power the radio first, wait 10 seconds (important), then power the receiver, wait for the connect, don't do anything else. Does it connect eventually within 30-60 seconds or does not connect at all and the receiver goes to WiFi mode after 60s.

And what about this?

closed 2024-11-17 by druckgott