ExpressLRS/ExpressLRS #3038: Compiling errors on macOS Ventura 13.7

etalot, 2024-11-16:
## Current Behavior

Compiling errors on macOS Ventura 13.7:

```
src/rx-serial/SerialIO.cpp:12:17: error: use of undeclared identifier 'min'; did you mean 'fmin'?

src/rx-serial/SerialCRSF.cpp:95:47: error: use of undeclared identifier 'map'; did you mean 'fmap'?

src/rx-serial/SerialIO.cpp:13:17: error: no member named 'readBytes' in 'Stream'

src/rxtx_common.cpp:7:10: fatal error: 'Wire.h' file not found
```

## Steps to Reproduce

pio run

## Details

target: native

Please help

mha1, 2024-11-16:
Why would you choose target "native"? What are you trying to do?

etalot, 2024-11-17:
> Why would you choose target "native"? What are you trying to do?

I do not choose. There are no any instruction in README or DOC how to compile or choosing target. File "platformio.ini" with data already exist in project. So I use Platformio doc - "pio run".

pkendall64, 2024-11-17:
you should use `pio run -e <environment>` e.g
```
pio run -e Unified_ESP32_2400_RX_via_UART
```