ExpressLRS/ExpressLRS #3042: CRSF not working in v3.5.2

Quarmire, 2024-11-18:
## Current Behavior

I am using a H7 Kakute Mini (w/ NAND) flight controller with a RadioMaster RP3 Diversity reciever.  My transmitter is a RadioMaster Pocket ELRS 2.4G.  Both RX and TX are flashed to 3.5.2 but CRSF gives no sign of life when checking the QGroundControl or MissionPlanner Radio tab.  Switching to SBUS allows the flight controller to receiver RC inputs.  After some hours of troubleshooting, a downgrade to v3.3.0 allowed me to use CRSF again.

## Steps to Reproduce
Flash ELRS v3.5.2 on RX and TX (using binding phrase).
Connect FC to PC via USB (serial).
Bind radios and move joysticks.
Use GCS to monitor radio input (SBUS vs CRSF).

## Possible Solution
I downgraded both RX and TX to v3.3.0, and CRSF just works.

## Details
The TX binds with the RX properly but it is the segment between the FC and the RX that is the issue.  Wires are soldered correctly, and baud rates are matched properly.

Switching to SBUS works, but CRSF has all the useful telemetry data.  I'd also like to try the new MAVLink features of ELRS so I don't want to be stuck on 3.3.0.

## Your Environment
- TX hardware: RadioMaster Pocket ELRS 2.4G
- RX hardware: RadioMaster RP3 Diversity
- EdgeTX version: edgetx-pocket v2.10.5
- ExpressLRS version: 3.5.2
- Telemetry Ratio: Std (1:128)

pkendall64, 2024-11-18:
If it works when you select SBUS then you either have it wired to the SBUS pad on the FC or some other configuration is not quite right. You would be better asking on discord in the #help-and-support channel as this is not a software bug.

pkendall64, 2024-11-18:
Closing this ticket as I see that @deadbytefpv sorted you out on discord.

closed 2024-11-18 by pkendall64