ExpressLRS/ExpressLRS #3046: [Mavlink] Stubborn Sender feature breaks Message sequences on 900Mhz RC Link

menschel, 2024-11-21:
## Current Behavior

Using QGroundControl 4.4.2 together with 900Mhz Mavlink via ELRS feature yields Sequence Issues on Mission Plans, i.e. Waypoints and Rallypoints.

Refer to original Issue for details
https://github.com/mavlink/qgroundcontrol/issues/12076

It boils down to QGC asking for a waypoint and receiving something other in response.
This is most likely a race condition related to https://www.expresslrs.org/software/mavlink/#stubborn-sender .

Same setup with a 2G4 setup works without issues.

## Steps to Reproduce

1. Start QGC on a device that is setup to provide an Access Point and the ability to create a UDP socket on port 14550.

2. Start a 900Mhz ELRS based RC which can connect to the Access Point and is setup to provide a mavlink connection via backpack. Note: The whole thing is also present if USB-Serial is used, so no relation to backpack.

3. Turn on the Plane / Copter or whatever system is controlled by the radio and provides the Mavlink Stream via RC.

4. Watch QGC start the connection, progress until ~50% and then fail with an Error Message `Mission read failed, maximum retries exceeded`.

## Possible Solution (Not obligatory)

Disable Stubborn Sender Feature for testing. Not verified!

## Details

It is impossible to use QGC with Mavlink from ELRS for mission planning or auto missions (900Mhz).
The exact same system but using the RM TX12MK2 internal 2G4 module and a BetaFPV ELRS Lite Receiver 2G4 works without issues on the same scenario.
A DIY mLRS setup on 868Mhz using Seeed WIO-E5 hardware works perfectly with exceptionally good performance on the RC link.

## Your Environment

- TX hardware: RM Bandit Micro on 868Mhz
- RX hardware: BetaFPV SuperD 900 on 868Mhz
- Handset model: RM TX12MK2
- EdgeTX 2.10.5
- ExpressLRS 3.5.1 and 3.5.2, backpack 1.5.1
- Packet Rate: 200Hz
- Telemetry Ratio: 1:2
- user_defines: ordinary EU 868, AccessPoint Settings and nothing else.

JyeSmith, 2024-11-22:
This wont be a SS issue as I imagine you are testing on the bench and at close range.  So no, or very few, dropped packets and hence the SS wont actually be doing any retries.

This all works with Mission Planner, and by the sounds of it higher OTA data rates with QGC (2G4).  So Im thinking the response from the FC back to QGC is taking too long and QGC is timeout out.

Something ELRS can test and reducing the size of the flow control buffer that is used to hold data from the FC to send back to QGC.  If there are already bytes in the buffer, and QGC request waypoint info, then these new packets will be added to the buffer but behind the currently buffered packets.  If we reduce the buffer size then there may be fewer bytes held and in front of the requested waypoint info.  But we may also run into other flow control issue and this all needs testing.

If you would like to test this yourself this is the buffer that could be reduced.

https://github.com/ExpressLRS/ExpressLRS/blob/126f3f902bfae1768bddc8dba54600b3f0b37ee6/src/src/rx-serial/SerialMavlink.h#L5

On the QGC side of this, if there is a timeout, could it be extended to account for low data rate links?

menschel, 2024-11-22:
I will investigate both options and report back.

menschel, 2024-11-22:
Tested option 1 with MAV_INPUT_BUF_LEN 512 on both TX and RX. No Error Messages but performance is sub par. I'll flash ardupilot 4.6.0beta1 and see if it's any better.

closed 2024-12-10 by menschel

JyeSmith, 2024-12-10:
@menschel thanks a heap for testing and confirming a hunch.  Are the new values in the PR the same as that used in Mission Planner?

It makes sense that 2G4 will be faster since it has a higher B/s, so it will take 900M longer to clear the internal ELRS buffer and finally transmit the ack packet.

menschel, 2024-12-10:
Honestly I have not checked how Mission Planner does things, because I was not able to get the only Windows PC that I have left, a gaming rig with Windows 10 Home Edition, to open up port 14550. Therefore the whole test setup never worked out on Windows.