ExpressLRS/ExpressLRS #3053: expressLRS Configurator V1.7.7 can not connect st link

AminZallaghi, 2024-11-25:
I tried to update Frsky R9m Lite pro via expresslrs configurator v1.7.7 but i have this error

Traceback (most recent call last):
  File "runpy.py", line 196, in _run_module_as_main

  File "runpy.py", line 86, in _run_code
  File "C:\Users\Amin Zallaghi\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\2909ba2db2909b5c38a299eed575082f2894baf1\firmware\flasher.pyz\__main__.py", line 3, in <module>
    if __name__ == '__main__':
  File "C:\Users\Amin Zallaghi\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\2909ba2db2909b5c38a299eed575082f2894baf1\firmware\flasher.pyz\_bootstrap\__init__.py", line 253, in bootstrap
  File "C:\Users\Amin Zallaghi\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\2909ba2db2909b5c38a299eed575082f2894baf1\firmware\flasher.pyz\_bootstrap\__init__.py", line 38, in run

  File "C:\Users\Amin Zallaghi\.shiv\flasher.pyz_39e2408ec3c41274ed588c24948dea8fb3fea8388981fd8c2a2d020b3d120cd6\site-packages\binary_configurator.py", line 419, in main
    return binary_flash.upload(options, args)
  File "C:\Users\Amin Zallaghi\.shiv\flasher.pyz_39e2408ec3c41274ed588c24948dea8fb3fea8388981fd8c2a2d020b3d120cd6\site-packages\binary_flash.py", line 198, in upload
    return upload_stm32_stlink(args, options)
  File "C:\Users\Amin Zallaghi\.shiv\flasher.pyz_39e2408ec3c41274ed588c24948dea8fb3fea8388981fd8c2a2d020b3d120cd6\site-packages\binary_flash.py", line 55, in upload_stm32_stlink

    stlink = external.pystlink.PyStlink(verbosity=1)
  File "C:\Users\Amin Zallaghi\.shiv\flasher.pyz_39e2408ec3c41274ed588c24948dea8fb3fea8388981fd8c2a2d020b3d120cd6\site-packages/external\pystlink\pystlink_api.py", line 42, in __init__
    self._connector = stlinkusb.StlinkUsbConnector(dbg=self._dbg, serial=self._serial, index=self._index)
  File "C:\Users\Amin Zallaghi\.shiv\flasher.pyz_39e2408ec3c41274ed588c24948dea8fb3fea8388981fd8c2a2d020b3d120cd6\site-packages/external\pystlink\lib\stlinkusb.py", line 95, in __init__
    raise stlinkex.StlinkException('ST-Link/V2 is not connected')
pystlink.lib.stlinkex.StlinkException: ST-Link/V2 is not connected

M-zhar, 2024-12-01:
Need proper code explanation.

AminZallaghi, 2024-12-06:
> Need proper code explanation.

how to send code explanation?

pkendall64, 2024-12-06:
I believe this error means that you need to install libusb.

AminZallaghi, 2024-12-06:
> I believe this error means that you need to install libusb.

I install it but problem not solved

pkendall64, 2024-12-29:
Could you try using the [web-flasher](https://expresslrs.github.io/web-flasher/)

closed 2025-03-31 by mha1