ExpressLRS/ExpressLRS #3063: ESP32-C3 receiver support

DrakoshaNarkosha, 2024-12-05:
## Current Behavior
After flashing empty ESP32-C3 i have next terminal output, the code does not start.
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
E (209) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='77a15e25'
Image checksum='fa4d5200'
[   210][E][esp32-hal-cpu.c:207] setCpuFrequencyMhz(): CPU clock could not be set to 240 MHz
```

## Steps to Reproduce

1. Take version 3.5.2 from repo.

2. Select Unified_ESP3C3_2400_RX_via_UART target.

3. Set -DRegulatory_Domain_US_433_WIDE.

4. Flash ESP32-C3.

## Possible Solution (Not obligatory)

Seems that some memory footprint is corrupted. Maybe it make sense to check partitions for C3.

## Your Environment
Bare ESP32-C3.

pkendall64, 2024-12-05:
Which ESP32C3 chip exactly? You need at least 4MB of flash. We have many ESP32C3 based receivers in the works and being released as well as development boards that have been used to test the firmware and have not run into this issue.

DrakoshaNarkosha, 2024-12-05:
[image]

DrakoshaNarkosha, 2024-12-05:
> Which ESP32C3 chip exactly? You need at least 4MB of flash. We have many ESP32C3 based receivers in the works and being released as well as development boards that have been used to test the firmware and have not run into this issue.

[image]

DrakoshaNarkosha, 2024-12-07:
After some research I found that GPIO2 and GPIO8 should be high during startup. But program still doesn`t run. I have next output:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
[   208][E][esp32-hal-cpu.c:207] setCpuFrequencyMhz(): CPU clock could not be set to 240 MHz
Config version 9
```

If I change clock frequency to 160MHz output is next:
```

pkendall64, 2024-12-12:
Why would you be flashing a 2400 target when you are selecting a subG domain? You need to be selecting `Unified_ESP32C3_900_RX_via_UART` if you have a SX1276 chip. Which looks like it given the power values you have in the target.

closed 2024-12-30 by pkendall64

DrakoshaNarkosha, 2025-01-06:
> Why would you be flashing a 2400 target when you are selecting a subG domain? You need to be selecting `Unified_ESP32C3_900_RX_via_UART` if you have a SX1276 chip. Which looks like it given the power values you have in the target.

Type of flashing target doesn`t matter in this case. If I remove "serial_rx" and "serial_tx" fields from hardware config file logs will work and no stop on "Config version 9" (it`s the last log output before pin initialization from the config file). But, communication with flight controller will not be present. To work with it "serial_rx" and "serial_tx" pins must be set in hardware config.