ExpressLRS/ExpressLRS #3064: Jumper T20 Gemini inturnal module TX

OldBuzzardFPV, 2024-12-06:
## upgraded to newer firmware

After upgrade from 3.4.3 Gemini to 3.5.3 on my Jumper T20S v1. And updated the lua script. I lost the functionality to connect to any of my Drones gemini or not.

## would not bind to anything. Could change output powers Fan setting but would not connect 

##  tried to go back to 3.4.3 in v1.7.7 and was still the same with loss of connection. 
Lucky I had an older firmware of 3.4.3 Gemini from a past update and it all came back. And started to connect again.

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Context

## Your Environment

- Operating System and version:
- TX hardware:
- RX hardware:

pkendall64, 2024-12-06:
Please ask on discord. It sounds like you have flashed the wrong firmware to your module!

OldBuzzardFPV, 2024-12-08:
There is only 1 T20 Gemini 2.4 on the pick list

pkendall64, 2024-12-23:
@OldBuzzardFPV did you get this sorted?

OldBuzzardFPV, 2024-12-25:
Yes problem repaired

OldBuzzardFPV, 2024-12-25:
Awesome, thanks for the update!