ExpressLRS/ExpressLRS #3074: [MAVLINK] Link mode = Mavlink breaks Yaapu telemetry script

menschel, 2024-12-19:
When selecting Link Mode = Mavlink, the telemetry is no longer forwarded to the handset and this breaks the [Yaapu](https://github.com/yaapu/FrskyTelemetryScript) telemetry script in the handset.

## Steps to Reproduce

Follow the guide for [Mavlink](https://www.expresslrs.org/software/mavlink/) and observe that Yaapu Telemetry in the Handset does not get any data.

## Possible Solution (Not obligatory)

IMO it's just a mission `handset->sendTelemetryToTX(CRSFinBuffer);` in this conditional branch.
https://github.com/ExpressLRS/ExpressLRS/blob/master/src/src/tx_main.cpp#L1435

Yes I can change, test and push PR but I'd like to ask before if this behavior is intended?

It does look like a rebase issue to me.

## Details
Yaapu reads Mavlink Telemetry embedded in CRSF and this is no longer sent when Link Mode is Mavlink.

## Your Environment

Independent of Hardware.

menschel, 2024-12-19:
I'll build from trunk just to verify I'm not pointing at something that is already fixed.

menschel, 2024-12-19:
I realize this is is now a feature request and not a bug.

menschel, 2024-12-26:
Solved by https://github.com/ExpressLRS/ExpressLRS/pull/3077

closed 2024-12-26 by menschel

pkendall64, 2025-07-27:
> Hi, was this actually fixed? I saw that the PR was merged in April, and my hardware is currently on 3.5.6. Yaapu still complains that it's not getting any telemetry, though it is getting GPS coordinates. 

It is not in the 3.x line, it is merged into the master branch which will become the 4.x line _when_ it is released.

Roasted715Jr, 2025-07-27:
Thanks for the help, flashed the master branch and it's working now :)

SvenIffert, 2025-08-11:
> Thanks for the help, flashed the master branch and it's working now :)

Could you please be more specific? What exactly did you flash? Thank you!

Roasted715Jr, 2025-10-30:
> > Thanks for the help, flashed the master branch and it's working now :)
> 
> Could you please be more specific? What exactly did you flash? Thank you!

Sorry, don't check my github notifications.

At the time, I just flashed what was on the master branch at that point. I selected the specific branch through the ELRS configurator.