ExpressLRS/ExpressLRS #3080: TX Power missing after switching to MAVlink and back to CRSF

dzaro-dev, 2024-12-25:
## Current Behavior

TX Power showing ZERO after switching back from MAVlink to CRSF.

## Steps to Reproduce

1. Follow the existing guide to set up MAVlink for INAV.

2. Switch back from MAVlink to CRSF

3. Look at the TX power in the OSD and see that it shows ZERO.

## Possible Solution (Not obligatory)

At first, I thought that this was related to INAV8, which now has some MAVlink support, but that was not the case.

I resolved the issue by reflashing both the transmitter and the receiver with ELRS 3.5.3. 

This cleared the fault and the TX power was showing again in the OSD.

NOTE: TX power was showing as a telemetry item on my radio.

## Details

Here is how the whole thing looked like on my Goggles 2 using the O3 Air Unit:

| Before setting up MAVlink | After reverting from MAVlink back to CRSF |
|--------|--------|
| [image] | [image] |

## Your Environment

- TX hardware: Radiomaster ELRS Micrco Module
- RX hardware: Matek ELRS R24-D
- Handset model: Radiomaster TX16S MK2
- EdgeTX: 2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.5.3
- Packet Rate: 150Hz
- Telemetry Ratio: 1/2
- user_defines: custom build, not relevant

kasatka60, 2025-01-06:
Same problem.
TX hardware: HappyModel ES900TX 
RX hardware: HappyModel ES900RX
ExpressLRS version (TX & RX MUST MATCH): 3.5.3
Handset model: Jumper t12 pro
EdgeTX: 2.10.5
Packet Rate: 50Hz
Telemetry Ratio: 1/2

kasatka60, 2025-01-22:
I'll try to play with it myself. Maybe I'll make it work.

MUSTARDTIGERFPV, 2025-03-24:
Since we've not been able to reproduce, I'll close this for now.

closed 2025-03-24 by MUSTARDTIGERFPV