ExpressLRS/ExpressLRS #3085: Ghost Atto - 3.5.x does not work properly or not at all

TheIsotopes, 2024-12-29:
## Current Behavior

Will testing Expresslrs on Ghost Atto, but it does not working. Imo the last version that works is 3.4.3.

with 3.5.0 and 3.5.1 the elrs script page is broken (see screenshot)

[image]

[image]

and with 3.5.2 and 3.5.3 the Ghost Atto will not starting correct. on boot no rainbow led effect only yellow and then it stuck with blue led and hangs.

## Steps to Reproduce

use any 3.5.x version on Ghost Atto 

## Possible Solution (Not obligatory)

will using a working 3.5.x version on Ghost Atto

## Details

## Your Environment

- TX hardware: intern RM TX16S MKII ELRS Module
- RX hardware: Ghost Atto
- Handset model: RM TX16 MKII
- EdgeTX version: 2.10.5 / 2.11 nightly
- ExpressLRS version (TX & RX MUST MATCH): 3.5.3 /3.4.3
- Packet Rate: 250 / 500
- Telemetry Ratio: Std
- user_defines:

pkendall64, 2024-12-29:
STM32 based TX modules and receivers have been removed from the codebase in `master` so will not be available in V4 onwards.
So I doubt that any of the devs have any incentive to fix this. You would be far better of getting an dedicated ELRS receiver from a manufacturer that is supported and also supports the development of ELRS.

TheIsotopes, 2024-12-30:
> ExpressLRS 3.5.x will be the last version to support STM32-based hardware. This includes the Happymodel PP, ES915 Tx and Rx, early NamimnoRC Flash and Voyager, FrSky R9, SIYI and ImmersionRC hardware.

Maybe we should take a closer look at this problem, otherwise the statement above is not really true.

pkendall64, 2024-12-30:
[image]
Flashed an ATTO this morning and it's all working correctly. Doing the rainbow start, flashing yellow till a connection is made then solid color, as expected. Lua loads correctly as well as can be seen in the screenshot.

TheIsotopes, 2024-12-31:
Can you explain which flasher you are used. I don't understand why it doesn't work for me!?
Web flasher / Configurator which version? Which Browser using with Web flasher? Installation Method - Betaflight Passthrough or STLink?

I don't understand it. :((

pkendall64, 2024-12-31:
I used the web-flasher in the Chrome browser on a Mac using the betaflight passthrough method.

TheIsotopes, 2024-12-31:
So with the web flasher I have no chance of installing anything, neither via stlink nor via betaflight passthrough, none of it works.

Could it be that something goes wrong when compiling and flashing in the offline flasher?

TheIsotopes, 2024-12-31:
may I guess? you are using the fcc version, correct?

I just found out that this problem only occurs with the lbt version. If I use the 3.5.3 fcc version everything is fine.

pkendall64, 2024-12-31:
Looks like a memory issue. i.e. we've added too many features for the Atto (in fact all STM32's) to handle.

pdxtigre, 2025-01-08:
Is it also true for FrSky R9M? I compiled the latest code from 3.x.x-maintenance branch as well as tag 3.5.3, neither worked for me if I compiled directly on my computer.  However, if I used **Configurator** with the released version 3.5.3 or git branch (also via configurator) then the FW worked.  Why is that? I really need to mod the code for my custom build not just some settings.

pkendall64, 2025-01-08:
> Is it also true for FrSky R9M? I compiled the latest code from 3.x.x-maintenance branch as well as tag 3.5.3, neither worked for me if I compiled directly on my computer. However, if I used [...]

Yes it will likely be true for all STM32 targets.
You can base you changes on the PR linked in this ticket which removes the MAVLink code, which was effectively disabled before, and that should reduce the memory footprint enough to work again.

pdxtigre, 2025-01-08:
Thanks Kendall, I will try that and test again.  I wonder how the Configurator would work differently.

SnowLeopardFPV, 2025-01-16:
I've just spent the last 2-3 hours scratching my head and trying to troubleshoot why my Ghost Atto wasn't binding after updating it to ELRS 3.5.3 from ELRS 3.4.3, then I came across this issue thread. I've tried various combinations of ELRS 3.x firmware on the TX (Namimno Flash OLED) and the Atto to try and figure out which combination was breaking something to no avail. After flashing any version of 3.5.x to the Atto and NOT power cycling it I can get a successful bind, but after a power cycle I then start experiencing unexpected issues. I'm doing a full chip erase each time I flash new firmware and I am using the FCC version. I have tried using both the web flasher and the offline ELRS Configurator but just get the same results:
- 3.5.0 -> After power cycling the Atto, the RGB LED on it does the rainbow colour cycling and then starts flashing yellow. After this, sometimes I can bind and sometimes not. It seems to be inconsistent and unreliable.
- 3.5.1 -> After power cycling the Atto, the RGB LED on it initially turns a solid greeny-yellow colour, then it turns a solid blue colour and gets stuck there. There is no rainbow cycling and it never flashes yellow, so there is no way to bind to it.
- 3.5.2 & 3.5.3 -> After power cycling the Atto, the RGB LED on it does the rainbow colour cycling then flashes yellow, but binding is never successful.

Even though 3.5.0 sometimes works, I don't trust running that on my Atto with the intermittent binding issue, so I've now flashed it back to 3.4.3 again (with my Namimno Flash OLED TX on 3.5.3) which is working as expected with no issues. Based on what has been mentioned in this thread, I guess I will either have to live with ELRS 3.4.3 version on the Atto forever, or retire it and replace it with a newer ESP32 based one.

pkendall64, 2025-01-16:
@SnowLeopardFPV I have a fix already merged into 3.x.x-maintenance branch and will be included in an upcoming 3.5.4 if you would like to test it.
Also be aware that 3.5 is the last release that will support STM32 for all ELRS devices. They just don't have enough RAM for FLASH for the features that are being added. So you would be far better getting some ESP based receivers so they are supported in future releases. Look at the radiomaster XR series or others that use the LR1121 for real future proofing.

SnowLeopardFPV, 2025-01-16:
@pkendall64 Thanks for the reply. I've just flashed the 3.x.x-maintenance branch to my Atto using the offline Configurator and can report back that the Atto is now binding to my Namimno Flash OLED TX using the maintenance release firmware, and does so even after a power cycle. So the binding issue appears to be fixed for the upcoming 3.5.4 release. I won't be able to test it in anger by doing any flights for some time, but I'll do so as soon as I get a chance. In the meantime I will keep an eye out for the official 3.5.4 release and flash that when available. I like to try and keep all of my ELRS receivers on the same version and the Atto in one of my quads was the only problematic one, so thank you for your efforts in getting this particular issue fixed.

Once ELRS 4.x is released I will likely just replace the Atto with a Radiomaster ELRS receiver, but for now I will leave the Atto installed in the quad while it's still able to work fine on the latest ELRS 3.x version.

pkendall64, 2025-01-17:
@SnowLeopardFPV Thank you very much for reporting back.

PR #3088 (2025-01-11): No MAVLink for you STM32!
MAVLink on STM32 was effectively disabled by having a stub implementation, but... users could still select MAVLink in the Lua script and be confused as to why it's not working. This PR removes it completely.
Also, some of the big-fat buffers were still allocated and reduced the available memory and cause issues for memory limited STM32 devices and they would also crash/fail to start!

Fixes #3085
closed 2025-03-24 by MUSTARDTIGERFPV