ExpressLRS/ExpressLRS #3092: elrsV3.lua Script Syntax Error

afrigeri, 2025-01-04:
## Current Behavior

ELRS v3.3.0 on Radiomaster Micro mounted on X9D+ (EdgeTX 2.10.5), I'm not able to access ELRS configuration, because I got the error:

```
Scrpt Syntax error
TOOLS/elrsV3.lua:562
syntax error near char(1)
```

## Steps to Reproduce

1. Flash ELRS v3.3.0 on Radiomaster Micro module

2. Copy elrsv3.lua into TOOLS folder in the X9D+ sdcard

3. configure model with external module CRSF / 921k / 500Hz.  The module's led lights up.

4.  long press menu, TOOLS -> ExpressLRS gives syntax error

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: Taranis X9D+
- RX hardware: 
- Handset model:
- EdgeTX version 2.10.5
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0 
- Packet Rate: 921k
- Telemetry Ratio:
- user_defines:

pkendall64, 2025-01-04:
This is a known issue with the X9D+ (most Frsky radios). They have a slow inverter and cannot operate at the higher baud rates.  You can potentially try the ONEBIT sampling mode in EdgeTX and see if that helps. https://www.expresslrs.org/hardware/x9d-troubleshooting/?h=onebit#troubleshoot
Other than that, theres not a lot that can be done other than to suggest a better more modern radio.

closed 2025-01-04 by pkendall64

afrigeri, 2025-01-04:
Thanks for the info @pkendall64!   I removed / reinserted the battery plug and got rid of this but got into 'not enough memory' message described #2052. One of the solutions reported is the ONEBIT sampling mode you suggested.