ExpressLRS/ExpressLRS #3099: Issues with Dual-Band Receivers in Normal Binding Mode

zyren, 2025-01-10:
## Current Behavior

## Steps to Reproduce

Using a Dual Band transmitter and Dual Band receiver, both with firmware version 3.5.3, some issues were encountered during normal binding mode. We hope these can be resolved.

---

### Binding in 2.4GHz Mode:

1. Select a **Packet Rate** of **500Hz 2.4GHz**.
2. Hold down the button on the receiver for 10 seconds to reset it and enter binding mode. *(Note: Be sure to reset the receiver; otherwise, it may lead to misleading behavior.)*
3. Press **Bind** on the transmitter. After waiting for a few seconds, the connection is successfully established.
4. Turn off the receiver's power directly. Upon powering it on again, it takes a few seconds to reconnect instead of connecting immediately.
5. Repeat steps 1–3 above, but this time, without turning off the receiver's power, change the **Packet Rate** to another value.
6. Turn off the receiver’s power, then turn it back on. Now, it connects immediately.
7. Repeat steps 1–3 above, but instead of powering off the receiver, restart the transmitter (or remove the transmitter’s battery and restart it). The receiver connects immediately.
8. Turn off the receiver’s power and then power it back on. Again, it connects immediately.
9. Repeat steps 1–3 and step 7. First, turn off the transmitter, then turn off the receiver. Turn on the transmitter again, followed by the receiver. The connection is established immediately.

---

### Binding in Low Band Mode:

1. Select a **Packet Rate** of **200Hz Low Band** and rebind.
2. Hold down the button on the receiver for 10 seconds to reset it and enter binding mode. *(Note: Be sure to reset the receiver; otherwise, it may lead to misleading behavior.)*
3. Press **Bind** on the transmitter. This time, the wait is approximately three times longer than in 2.4GHz mode (about 30–40 seconds) before the connection is successfully established.
4. Turn off the receiver’s power directly. Upon powering it on again, it connects immediately.

---

### Observations with Debug_Log Enabled and CRSF Signal Disabled on UART:

1. After binding in **2.4GHz mode**, the receiver does not store the Rate information during the first connection. As a result, each time the receiver is powered on, it cycles through five different Rates, causing a delay of several seconds before it connects to the transmitter.
2. After binding in **2.4GHz mode** and then switching to another Rate, the receiver does not cycle through the five Rates upon rebooting. Instead, it connects immediately.
3. After binding in **2.4GHz mode**, turning off the transmitter and restarting it (regardless of whether the receiver is restarted) prevents the receiver from cycling through the five Rates. It connects immediately.
4. After binding in **Low Band mode**, the receiver cycles through 15 different Rates during the binding process. This results in a much longer wait time (30–40 seconds) before binding is successful. However, after binding is complete, the receiver no longer cycles through the Rates during reboot, and it connects immediately.
5.  The DK500 and K1000 were also tested, and the results were exactly the same as those in 2.4GHz mode.

---

Additionally, I also tested the RP series 2.4GHz receivers in 2.4GHz mode. On the surface, they appear to connect very quickly. However, by enabling Debug_log for observation, it was revealed that the same phenomenon occurs. The difference is that the process is much faster, making it difficult to detect.

---

## Possible Solution (Not obligatory)

So the conclusion is that, after binding in 2.4GHz mode, the receiver does not save the current rate during the **first binding attempt**, and only the first attempt. If the rate is not switched or the transmitter is not turned off to trigger the receiver to save the current rate, the receiver will poll through each rate upon every reboot before successfully connecting. Once the rate is saved—either by switching the rate or by turning off the transmitter—future reboots of the receiver will result in an immediate connection with no waiting time.

Unless the user happens to change the rate or turns off the transmitter at least once before powering off the receiver, this issue will persist. However, many users find themselves stuck in this situation for a long time. They may have never changed the rate or turned off the transmitter before powering off the receiver, leading to constant complaints about why the receiver connects so slowly.

Although this does not seem like a major issue, it has caused significant confusion and inconvenience for many users during everyday use. We hope this issue can be resolved.

## Your Environment

- TX hardware: Nomad, GX12 Int-Module
- RX hardware: XR1, XR2, XR3, XR4, RP1, RP2 ...
- Handset model: Boxer, Pocket, GX12
- EdgeTX version: 2.10.5 
- ExpressLRS version (TX & RX MUST MATCH): 3.5.3
- Packet Rate: 50-500Hz 2.4Ghz, 50-200Hz Low Band
- Telemetry Ratio: Any
- user_defines: -DRegulatory_Domain_FCC_915

pkendall64, 2025-01-10:
This is all explained here https://www.expresslrs.org/info/init-rate/
The issue is that we cannot save the packet rate when a connection is initially made, as saving the rate to eeprom causes the link to be dropped because eeprom writes take quite some time to complete. And the code cannot run while saving to eeprom.

zyren, 2025-01-11:
Thank you very much for your explanation, I understand now.

closed 2025-01-24 by pkendall64