ExpressLRS/ExpressLRS #3101: Long time to regain signal?

Johanl1964, 2025-01-11:
Maybe not a bug, but a question... Well, I ran a ground test to check the range with GemX dualband, I drove until I lost the signal completely and got FS, but when I turned back it took a long time to get the signal back, even though I previously had a signal on the current route! So obviously very slow to regain the signal after it is completely lost! Can add that in case of short-term signal loss/FS, there is no delay at all!

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment. Ground test with a lot of radio shadow/obstacles

- TX hardware: GEPRC RX as TX 868
- RX hardware: GEPRC RX 868
- Handset model: Jumper T-Pro v2
- OpenTX version ETX Nightly
- ExpressLRS version Master 4.0.0
- Packet Rate: 100hz full
- Telemetry Ratio: 1:4
- user_defines: X-Band mode and Gemini, 10mW TX power, 50mW RX

Johanl1964, 2025-01-11:
Here we have examples of the road I drove and the signal, the conditions were the same in both directions, but it took a long time/distance to regain the signal!

[image]

pkendall64, 2025-01-17:
Do you have "Lock on first connection" enabled. This is the default and once a connection is made from the TX to the RX, the RX will not use any other packet rate until it's told to by the TX (via a Lua command) or if it's rebooted. If it rebooted then it will search around all the packet rates for a TX with a patching UID (bind-phrase) sending on one of the packet rates.

Johanl1964, 2025-01-17:
> Do you have "Lock on first connection" enabled. This is the default and once a connection is made from the TX to the RX, the RX will not use any other packet rate until it's told to by the TX (via a [...]

Yes, this is enabled! I did this test again with RM DBR4 on the same route, same conditions and settings, but DBR4 was even slower to regain the signal! So it's about waiting many minutes to regain the signal even though I demonstrably had a signal during the current route before the signal was lost

dingrgh, 2025-02-06:
Where in the vehicle did you place your equipment? On the roof? On a seat? Dash? It is possible that the signal is attenutated more in some orientations with repsect to the vehicle body (e.g. through the engine block, wide roof pillars...)

Try this.... Drive out about, say, 80% of the way out where you still have signal lock. Then turn around and see if the signal is lost. If so, it would indicate that the vehicle orientation with respect to your home point is a factor.

closed 2025-03-24 by MUSTARDTIGERFPV