ExpressLRS/ExpressLRS #3104: fatal error: stdint.h: No such file or directory

blackdoctor1, 2025-01-15:
## Current Behavior

Compiling .pio/build/Unified_ESP32C3_LR1121_RX_via_WIFI/src/common.cpp.o
In file included from /home/andrew/.platformio/packages/framework-arduinoespressif32@3.20011.230801/cores/esp32/Arduino.h:24,
                 from include/targets.h:3,
                 from include/common.h:4,
                 from src/common.cpp:1:
/home/andrew/.platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/lib/gcc/riscv32-esp-elf/8.4.0/include/stdint.h:9:15: fatal error: stdint.h: No such file or directory

blackdoctor1, 2025-01-15:
when trying to build a project in the target:
unified_esp32c3_lr1121_rx_via_wifi cannot find the corresponding file.

blackdoctor1, 2025-01-15:
Processing Unified_ESP32C3_LR1121_RX_via_UART (platform: espressif32@6.4.0; board: esp32-c3-devkitm-1; framework: arduino)
--------------------------------------------------------------------------------

Library Manager: Installing ottowinter/ESPAsyncWebServer-esphome @ 3.0.0

Unpacking 0%
 10%
 20%
 30%
 40%
 50%
 60%
 70%
 80%
 90%
 100%

Library Manager: ESPAsyncWebServer-esphome@3.0.0 has been installed!

Library Manager: Resolving dependencies...
Library Manager: Installing esphome/AsyncTCP-esphome

Unpacking 0%
 10% 20% 30% 40% 50%
 60% 70% 80%
 90% 100%

Library Manager: AsyncTCP-esphome@2.1.4 has been installed!

Library Manager: Installing esphome/AsyncTCP-esphome @ 2.0.1

Unpacking 0%
 10% 20% 30% 40% 50%
 60% 70% 80%
 90% 100%

Library Manager: AsyncTCP-esphome@2.0.1 has been installed!

Library Manager: Installing bblanchon/ArduinoJson @ 7.0.4

Unpacking 0%
 10%
 20%
 30%
 40%
 50%
 60%
 70%
 80%
 90%
 100%

Library Manager: ArduinoJson@7.0.4 has been installed!

Library Manager: Installing git+https://github.com/mavlink/c_library_v2.git#e54a8d2e8cf7985e689ad1c8c8f37dc0800ea87b

git version 2.43.0

Клонирование в «/home/andrew/.platformio/.cache/tmp/pkg-installing-rxj1b6hl»...

Указатель HEAD сейчас на коммите e54a8d2e autogenerated headers for rev https://github.com/mavlink/mavlink/tree/a3558d6b335d930fc01816fd168d16b3f38ed434

Library Manager: c_library_v2@0.0.0+20250115174012.sha.e54a8d2e has been installed!

Verbose mode can be enabled via `-v, --verbose` option

UID bytes: 255,189,73,53,7,66

***
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
[... 110 lines cut ...]

                 from src/rx-serial/SerialAirPort.h:1,
                 from src/rx-serial/SerialAirPort.cpp:3:
/home/andrew/.platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/lib/gcc/riscv32-esp-elf/8.4.0/include/stdint.h:9:15: fatal error: stdint.h: No such file or directory

****************************************************************
* Looking for stdint.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:stdint.h"
* Web  > https://registry.platformio.org/search?q=header:stdint.h
*
****************************************************************

 #include_next <stdint.h>
               ^~~~~~~~~~
compilation terminated.

*** [.pio/build/Unified_ESP32C3_LR1121_RX_via_UART/src/rx-serial/SerialAirPort.cpp.o] Error 1

========================= [FAILED] Took 16.99 seconds =========================

Environment                         Status    Duration
----------------------------------  --------  ------------
Unified_ESP32C3_LR1121_RX_via_UART  FAILED    00:00:16.988
==================== 1 failed, 0 succeeded in 00:00:16.988 ====================

mha1, 2025-01-15:
can you check if this folder exists and if so is it empty or is just stdint.h missing: `/home/andrew/.platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/lib/gcc/riscv32-esp-elf/8.4.0/include`

pkendall64, 2025-01-17:
I would clean out the data in your .platformio directory as it appears to be incomplete.

blackdoctor1, 2025-01-28:
> I would clean out the data in your .platformio directory as it appears to be incomplete.

senks. It`s works.

closed 2025-01-28 by pkendall64