ExpressLRS/ExpressLRS #3144: Bluetooth Gamepad doesn't work on android after 3.5.3 changes

mikuslaw, 2025-02-14:
## Current Behavior

On newest version of EdgeTX 2.10.6 on RadioMaster Pocket ELRS, on ELRS version 3.5.3 Android doesn't detect Bluetooth Gamepad (not listed when scanning for bt devices to pair). When ELRS version on TX is downgraded to 3.5.0, BT gamepad is detected. 

Possibly connected with commit 7ac5f4d

## Steps to Reproduce
1. Upgrade to v3.5.3
2. Start bluetooth gamepad on in ELRS Lua script on TX
3. Open bluetooth settings to scan for devices nearby
4. Wait for device to show up on the list of found BT devices.
5. List doesn't contain the Express LRS Gamepad

## Possible Solution (Not obligatory)

Possibly we could switch modes of BT based on setting in EdgeTX between Joystick, Multiaxis, Gamepad to report properly on BT.
## Details

## Your Environment

- TX hardware: Radiomaster Pocket ELRS Edge TX 2.10.6
- ExpressLRS version (TX & RX MUST MATCH): 3.5.3 and 3.5.0 (works)
- Android Phone: Samsung S10

mikuslaw, 2025-02-16:
Adding authors of the commit: @JyeSmith @MNS26 

It seems that there are some cases where reporting as joystick is better, and some cases where gamepad is necessary (android). Where did the reporting as a joystick worked better?

MNS26, 2025-02-16:
what? android doenst support joysticks?

MNS26, 2025-02-16:
ill look into that one... why would you even connect it as a controller to android though

mikuslaw, 2025-02-16:
There is orqa fpv.skydive simulator, i wanted to try it. Its not very good, but flyable. 

Keep in mind that S10 official build is Android 12, so they might have added support since then. Still it would be nice to make it configurable or stick with broadly supported mode. Do you remember why joystic mode is set in 3.5.3?

MNS26, 2025-02-16:
> Possibly we could switch modes of BT based on setting in EdgeTX between Joystick, Multiaxis, Gamepad to report properly on BT.

**do NOT report as "Multiaxis" its meant mostly for a 3d mouse**

err this would mean a considerable rewrite of the existing HID report code...
id be down for it i have plenty of experience with writing HID (yay?)
the issue i run into is persistence for the option, ive so far never managed to add settings to the configs

in theory it should also be possible to report as both a joystick and gamepad. I dont know how or if android supports multi device reports in a single descriptor, Linux doesnt.

MNS26, 2025-02-16:
> Wait for device to show up on the list of found BT devices.

does your PC see it? i can see my pi picoW thats reporting as a "joystick" fine from my phone

mikuslaw, 2025-02-16:
Bbbbb b bbvbbvb bbbb bbbbbbbbb bbbbbbbbbb  bbbbbb  bbvbbbbb bbbvbb bbbbbb
bbbvbbbbvbbbbbb bbbbbbbb bbbbbbvbbbbb  bbbbbb bbvvbbb b b bbb bbvb bb  bb b
bbbv b bbbbbvbbbvbbv bb bbbv bbb  bbvbbbvb bvbb bb bb  vb bvb vvbbb bbbbbbb
bbbbbb b bbb nv  bb  bb vbbbbb bb b  b bbbbbbbvbvbb vbb b vv   v bb
bbbbbbbbv bbbbb b  b bbbvb b bbvbbbbbbbbbbbbbbb  bbbbnbb bbbbb b bvbbb bbbb
bbbb bbbbb b b   bvbb bbbbbbbbb bb bb b   bbbbb bbbb  bbbbbbbbbb bvb bbvv b
bbbbbbbbbbbb  bb b  bbv  bbbb  b  bbbbbbn bbb bbbb b b bbbbbb bb bbbbbbbbvv
b bbvbbb  vbbb bv b bbbbbbbbbbb bbnb vv bbbbb bb  bbbbbv bb kbbbbb bbbbbb
bbb v  bbbbvvbv bnbbbbb b bb bvbvbbbvbbvb bbbvvvvbb bbbbb bbbbbbbb  b vbvbb
vb  bkbnbbb bb b  vv b bbbbbbvbbbbbbb   bbbvbb b bbnvbbbb bbbbb b
bvbbbbbbbbb  b  b bb vbbbb bb bb  bbbbbbbbbbbbbbb  bbbbbbb bbb bbbbbbbb
bb   bb b bbbbbbbvbb  vbbbbbb bvbbb  vbbvbbb bbbvbbvb vbbbbbbbbbnbbvbbbb
vbbbbb bb bbb bbbbbbvb  b bbbbbb b b b  bb bbvb bvbbbbbb b bbbb bbnbbbbbb b
bb vb bbv bbbbbbbb bbbbbb  bbbbb b b  bv bb b bbbb b bbbbbbb bbvbb bbvbvb
bbb bb vbb nbv vb bbn vbbbb bbbnb  bbb bbbb b  bb    v b  b   bbbv b

mikuslaw, 2025-02-16:
3.5.3 is seen by windows, but not S10. 3.5.0 is seem by android, didnt check yet on windows.

mikuslaw, 2025-02-16:
Also, that was just a guess that the commit was connected with the change in behavior i observed. I will try (tomorow after work) if i can use elrs configurator to build 3.5.3 with revert of that line and see if that isolates the issue.

mikuslaw, 2025-02-17:
Hi, I flashed again with 3.5.3, tested on android 14 and then again on android 12 (S10). Not sure why, but now it works fully as expected. I can't put my finger on it why it didn't work last time. Either there was something stuck in the esp bt implementation, that reboot+flash to 3.5.0 cleared (and I attributed that to version change), or my test setup was wrong (maybe other device was already connecting and esp was not advertising any more).

Thank you very much for your help and sorry for wasting your time. We can close this one.

closed 2025-02-17 by mikuslaw