ExpressLRS/ExpressLRS #3155: Signal lost, failsafe with no signal recovery with ESP8285 Rx as Tx / LBT

Steini63, 2025-03-09:
## Current Behavior

I had two crashes due to signal loss with an electric glider. I did a lot of research into the cause and have now been able to reproduce and localize the error on the bench. 

The error pattern is identical to [ISSUE #2905](https://github.com/ExpressLRS/ExpressLRS/issues/2905), the differences are:
- Firmware V 3.5.3
- Receiver Cyclone 7Ch PWM
- Tx modules ESP8285 2.4GHz RX as TX

## Steps to Reproduce / Details

My findings:

1. Telemetry and control signal are affected.
2. The receiver goes into failsafe mode.
3. The transmitter module can still be addressed via LUA script.
4. The signal loss remains until the transmitter module is restarted.
5. After the restart, the signal to the receiver and the telemetry are immediately available again.
6. I have tried 4 different receivers (as Tx); the signal is retained for different lengths of time, but it always happens at some point.
7. I have tested 3 different firmwares (Generic ESP8285 Full-duplex 2.4GHz RX as TX, BETAFPV 2.4GHz Nano TX (as Tx), SpeedyBee Nano 2.4GHz RX (as Tx)); no differences.
8. The time to signal loss is significantly shorter when another transmitter is in operation (in my tests = FlySky AFHDS V1).
9. The error only occurs with LBT firmware, with ISM firmware I have carried out 8-hour tests without any problems.
10. Tx modules with ISM firmware also bind with an LBT receiver.

Could it be that the "ESP8285 2.4GHz RX as TX" firmwares continue to have problems with LBT? Have the LBT changes from V 3.4.0 also been incorporated into the "Rx as Tx" code?

Thanks for all the work to ELRS.

Frank

## Your Environment

- TX hardware:  DIY Module in module bay with ESP8285 2.4GHz RX as TX.
- RX hardware: Receiver Cyclone 7Ch PWM
- Handset model: Tx16S
- EdgeTx 2.10.3 and 2.11 rc2
- ExpressLRS version (TX & RX MUST MATCH):  3.5.3
- Packet Rate: 333Hz Full
- Telemetry Ratio: 1:8
- user_defines: Binding phrase, WiFi startup delay 20 sec., Tx Power 100mW Dyn, Link-Mode normal, Model Match OFF

mha1, 2025-03-09:
1. All LBT fixes are in the 3.5.3 code base, hence also in RX as TX
2. I'd never recommend an 8285 RX as TX
3. Cyclone is generally not supported by the ELRS Team, hence no checking schematics, no measurements, no functional testing, no approval and of course no Configurator entry. I bet Cyclone doesn't have a TCXO and most likely has bad/wrong load capacitors on the oscillator.

Due to the "not-supported" fact you are pretty much on your own on this one as we can't tell if the hardware is in fact built to ELRS HW requirements, hence being ELRS capable. Same goes for your DIY receiver.

Recommendation: bin Cyclone HW and get yourself a decent ELRS TX module

Steini63, 2025-03-09:
Thanks for the quick reply!

I totally understand the attitude of the ELRS developers towards companies like Cyclone and their behavior. I bought the receiver before I became aware of this discussion and will avoid their products in future.

The DIY part of my transmitter module only concerns the power supply and the inversion of the signals so that the module can be operated in the standard module slot. The crucial part consists of commercially available receivers (SpeedyBee Nano Rx, BetaFPV Nano Rx). 

My descriptions have hopefully shown that the error is most likely not to be found in the hardware. After all, the ISM version runs for hours without any problems. Using the LBT firmware without changing the hardware reproducibly leads to signal loss. I would also like to point out that such a loss of signal is the worst possible error for flying objects. 
However, I also understand that there are too many unknown factors in my constellation that make an exact analysis difficult. Hopefully the error will not occur with a decent ELRS TX module. I will continue testing and look forward to ELRS V4!

Best regards
Frank