ExpressLRS/ExpressLRS #3156: Enabling MAVLINK will change the returned CRSF altitude from altitude above sea level to relative altitude.

HuanHuanTuanZ, 2025-03-10:
During my attempt to write a program to read the transmitted data, I encountered this phenomenon, which has caused significant confusion in calculating the relative altitude between multiple devices (other devices only have GPS and no IMU, so they can only use altitude above sea level). I am unsure whether this is an oversight or an intentional design, as the ELRS source code is too complex for me to decipher.

HuanHuanTuanZ, 2025-03-10:
version：3.5.3

MakeSomeFakeNews, 2025-03-24:
inav 固件默认是相对高度。如果是ardupilot 。 也是有相对高度的。 需要你自己去解析 0x80那个包

MUSTARDTIGERFPV, 2025-03-24:
> During my attempt to write a program to read the transmitted data, I encountered this phenomenon, which has caused significant confusion in calculating the relative altitude between multiple devices [...]

https://github.com/ExpressLRS/ExpressLRS/blob/master/src/lib/MAVLink/MAVLink.cpp#L48

As MakeSomeFakeNews says, the behavior is different between platforms, so we offer it as a build-time flag if you really wish to change it.

closed 2025-03-24 by MUSTARDTIGERFPV