ExpressLRS/ExpressLRS #3157: Reconnect issue after failsafe in full res modes

agnauck, 2025-03-13:
## Current Behavior

After failsafe in a full res modes the receiver is sometimes not re-connecting correctly.
I can reproduce this very often with those settings:

Packet Rate: 100Hz
Telemetry ratio 1:8
Switch Mode 8ch

When the issue happens then the receiver is not re-connecting correctly. I does not establish telemetry. The servos are moving randomly between failsafe position and radio output. Sometimes the servos move, but very slow, and they are stuttering.

Sometimes it does not reconnect correctly at all, and sometimes it takes very long time until it reconnects.

Its looks like its half connected, or switches between connected and failsafe all the time.

## Steps to Reproduce

1) Setup radio and receiver with Packet rate 100Hz, Telemetry Ratio 1:8 and switch mode 8ch
2) failsafe the receiver by turning the transmitter off 
3) Turn the receiver back on see if the RX reconnect correctly

This issue happens **very often** for me, but **not always**. If you cannot reproduce immediately then try to turn off and on the TX many times.

## Details
I have seen this issue with other full res modes as well (12ch mixed and 16ch rate/2).
All full res modes seem to be affected.

The issue seems to be related to the telemetry ratio. The higher it is the easier it seems to reproducible.

- 2x Radiomaster TX16s MK2
- can reproduce with Betav Superp 14ch, Radiomaster ER8 and ER8GV
- EdgeTX 2.10.6
- ExpressLRS version 3.5.3 LBT
- Packet Rate: 00Hz ll
- Telemetry Ratio: 1:8
- Switch Mode 8ch

agnauck, 2025-03-13:
flashed FCC firmware on the TX and RX. Happens with FCC as well. So it does not seem to be a LBT issue.
Also, the higher the telemetry ratio the easier to reproduce.

pkendall64, 2025-03-13:
Could you test with this PR #3154 and see if it helps.

agnauck, 2025-03-13:
yes I can, should I install it on RX and TX?

pkendall64, 2025-03-13:
The code change is only on the TX. But you would want both on the same version.

pkendall64, 2025-03-13:
> I tested PR [#3154](https://github.com/ExpressLRS/ExpressLRS/pull/3154) Is does not resolve this issue

Ok, thanks for testing.

agnauck, 2025-03-13:
I did more testing.
The problem seems to exist on switch modes 8ch and 12ch mixed, 16ch rate/2 seems to work fine.

agnauck, 2025-03-14:
I did steps 1, 2 and 3. I stopped at 3 because resetting the tx config solved the issue.
Looks like the TX is causing the issue, not the RX.

Both configs are attached below.

I have a 2nd pretty much identical TX16s which still has the issue. I did not reset the tx-config on this radio yet in case we need more info from it to find the root cause.

Both radios were shipped with pretty outdated software. I usually install every new ELRS version immediately. So they both went through many ELS update cycles. 

[tx-config.json](https://github.com/user-attachments/files/19245410/tx-config.json)
[rx-config.json](https://github.com/user-attachments/files/19245411/rx-config.json)

imesita, 2025-03-15:
I have the same problem and it was solved by doing step 3 to clear the tx config but the problem comes back when I turn the band on from the vtx administrator settings and when I turn the band off the problem goes away so I assume this problem is caused by the backpack?

pkendall64, 2025-03-15:
> but the problem comes back when I turn the band on from the vtx administrator

This is a know problem if you are using an SPI based receiver on a FlightController

hawku, 2025-03-15:
I have replicated the VTX administrator bug with ELRS 3.5.3 on Zorro and BetaFPV UART receiver. This happens when the TX tries to send a VTX command to FC:
https://github.com/user-attachments/assets/40d883a6-ccaa-4426-8008-39e0b1c9749b

Can't replicate the same problem if the VTX administrator band is set to Off

imesita, 2025-03-15:
> > but the problem comes back when I turn the band on from the vtx administrator
> 
> This is a known problem if you are using an SPI based receiver on a FlightController, there is a fix for that in the 3.x.x-maintenance branch

I have a happymodel ep1 receiver.

agnauck, 2025-03-15:
I am flying sailplanes and helicopters. I don't use backpack and the VTX admin.
But I cannot rule out that I placed with the VTX admin in the Lua script a while ago, which then could have resulted in a broken tx config.

mha1, 2025-03-15:
> I have the same problem and it was solved by doing step 3 to clear the tx config but the problem comes back when I turn the band on from the vtx administrator settings and when I turn the band off the problem goes away so I assume this problem is caused by the backpack?

Can you please share more detail on what you do in the VTX Admin and Backpack settings

imesita, 2025-03-15:
> Can you please share more detail on what you do in the VTX Admin and Backpack settings

Below is a link to a video where you can see how the problem occurs.
https://www.youtube.com/watch?v=K7hXhvyRlmU

imesita, 2025-03-16:
I have ELRS version 3.5.2 on radiomaster gx12 backpack version 1.5.1 and 3.5.3 ELRS version on happymodel ep1.