ExpressLRS/ExpressLRS #3174: Dynamic power not correctly functioning with Nomad TX

a1axx, 2025-04-04:
## Current Behavior

Dynamic power with a Nomad Gemini TX appears to be too sensitive when used with LR1121 hw receivers

## Steps to Reproduce

1. Power on tx and rx

2. Move the tx away from the rx

3. Observe when a tx dynamic power-up happens

4. Move the tx closer to the rx again; dynamic power takes a long time to reduce or doesn't reduce at all.

## Possible Solution (Not obligatory)

Reconfigure power increase and decrease parameters

## Details

With sx1280 hw receivers, dynamic power is much less sensitive to power increases and decreases very quickly.

 
## Your Environment

Anywhere tested, indoors or outdoors in open fields.

- TX hardware: Nomad ELRS (using 2.4 only)
- RX hardware: DBR4, XR4, XR1 and speedybee 2.4
- Handset model: x9d
- Edgetx 
- ExpressLRS version (TX & RX MUST MATCH): tested with 3.5.3 and 3.x.x to 3.5.3 rx (as instructed to use 3.x.x)
- Packet Rate: 333
- Telemetry Ratio: 1:2
- user_defines:

robustini, 2025-04-07:
Encountered the same problem, with Ardupilot it does in fact occasionally engage failsafe, which does not happen with other modules and dynamic power.

mha1, 2025-04-07:
Yes, it does weird stuff. Here's a log of me just putting some 10-12m in-house between a Nomad and a BAYCKRC 100mW Gemini RX and going back. It increases power much to early and doesn't reduce it again.

[ELRSTest 868-2025-04-07-162520.csv](https://github.com/user-attachments/files/19634307/ELRSTest.868-2025-04-07-162520.csv)

[image]

While testing I had an even weirder situation where TPWR was increased to 1000mW but I only had set a max. power of 100mW.

mha1, 2025-04-08:
Please disregard the second part where TPWR supposedly went above the max. power setting. This was due to a stupid user error. My T15 was lying on my my workbench and playing with it led to unintentional "increase power" button presses on the Nomad. No problem here but maybe this raises the question if you should be able to increase the power beyond the max. power setting.

chris1seto, 2025-04-10:
I have also seen this issue on two different setups (TX16S + Nomad + XR4 or DBR4). Both with fully charged batteries.

SunjunKim, 2025-08-17:
Please try #3302 to see whether it fixes your problem.