ExpressLRS/ExpressLRS #3175: CW mode using CW.html to Evaluation Noise and antenna Performance

ltwin8, 2025-04-06:
## Current Behavior

HW: Radiomaster nomad v3.5.3

1. Enter WiFi Mode (STA OR AP) 
2. Connect to <IP>/cw.html
3. select given options (Radio 1 or 2) and Low band or not
4. click start cw
5. massure using tinySA 

In Most cases the CW did NOT start at all, some ties After multiple reloads it does. But when it does it only does if on one Setting.

E.g. it works on Radio 1 2440MHz then it only works After reload on Radio 1 at 2440.

## Steps to Reproduce

Follow Steps 1 through 5 at beginning.

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

Used multiple devices, Smartphones (iOS18), Mac (macOS13), Windows 10, Windows 11, tested in STA and AP Mode 
Cookies deleted, does not solve the issue.

Enter CW Mode to Evaluation the antenna Performance and Noise

## Your Environment

- TX hardware: Nomad
- RX hardware: Not relevant 
- Handset model: TX16S mkII or without (wifi per Button)
- OpenTX version (including nightly number): Not relevant
- ExpressLRS version (TX & RX MUST MATCH): 3.5.3
- Packet Rate: 250 (guess not relevant)
- Telemetry Ratio: std
- user_defines: none/Standard in configurator

pkendall64, 2025-04-07:
Works for me as I test devices on a near daily basis, and have no issues with it.

ltwin8, 2025-04-07:
Are you using it on a nomad?

On my HM 24TX (pro) HW it also works without any issue.

Is there any way to Test the PA?
To set specific PA amplification on CW Mode to meassure the Output.

pkendall64, 2025-04-07:
No, we set the power output to the 10mW minimum so as not to damage the PA while measuring the actual centre frequency to calculate the XTAL accuracy.
For testing the PA, we usually set the telemetry ratio to 1:2 and measure over a period of time with a spectrum analyser so we can get a good picture of the response across the entire band.