ExpressLRS/ExpressLRS #3178: XR4 RX noise causing GPS fix loss or reduced performance

a1axx, 2025-04-07:
## Current Behavior

Gps lock issues with Radiomaster XR receivers on 2.4GHz band

## Steps to Reproduce

1. Using an FC with a GPS on the 4.5V rail and Elrs on the 5V, power up USB and observe sats locked 

2. Power your transmitter on, configured to 2.4GHz 333Hz rate and 1:2 telem and power the FC from vbat to turn on the Elrs rx

3. Upon bind, satellites can reduce or lose lock entirely.

4. Reconfigure the tx to 200hz low band

5. Observe changes in gps performance 

6. Remove power from the Elrs rx and observe changes in gps performance 

Another test was using an immersion RC power meter with a 1.2ghz antenna.

Upon binding, a significant power is detected on the 1200MHz band. (Close to GPS) 

When reconfiguring to low band, the 1200MHz noise floor is significantly reduced.

Test was repeated with a DBR4 receiver. The 1200MHz noise was only observed for a second after reconfiguring the band.

## Details

Has caused a low quality lock resulting in a loss of GPS navigation in flight.

## Your Environment

- TX hardware: nomad 
- RX hardware: xr4, another xr4 and tested on a dbr4 (not affected in the same way)
- Handset model: x9d
- OpenTX version (including nightly number) Edgetx
- ExpressLRS version (TX & RX MUST MATCH): 3.5.3 rx and 3.x.x as instructed
- Packet Rate: 333
- Telemetry Ratio: 1:2
- user_defines:

tguillaume02, 2025-04-07:
I have the same issue with this setup: 
Tx: Nomad gemini
Rx: xr4
GPS: m8q

a1axx, 2025-04-08:
Performance is normal in 868 low band as tested last night. 22-25 satellites. Did not revert to 2.4 in flight for obvious reasons, could have broke the link and lost the lock.

lesykm, 2025-04-09:
Can repro.
Tested elrs receivers: Skystars SS24D, Matek R24S
Tested GPS modules: M8,M10
Firmwares 3.5.x (did not test 3.5.4 but I doubt it is any different).
Using mode: Mavlink/333Full

On weekend will try to measure interference on gps frequencies using TinySA and/or RtlSDR

Ayil911, 2025-04-11:
hey guys, I'm experiencing something similar, I have a GEPRC Xrossband ELRS rx and i have it set to X150 on the radio. Im on EU sub band 868mhz. I can't seem to get a GPS lock on my BZ-251 module whenever i'm using the Xcrossband. the GPS gets full 17-21 satelites when I use either 2.4ghz or 868mhz non Xcrossband. Radio is a RM TX16s Mk2 with RM Nomad module.
I've tested switching from normal 2.4ghz and Low band with good satelite reception to Xcrossband and the satelite count immediately drops and GPS position is unusable. I have also tried isolating the receiver by moving it away from the GPS module and also by covering the reciver with thick metal tape. Results are the same.
 kinda beats the purpose of me getting a Xcrossband RX.. lol. anyone with any ideas?

raul-ortega, 2025-04-11:
I have a similar issue in the GPS module Walksnail when using the 868 EU band with Radiomaster XR4. The wait time to reach even 8 satellites is very long. Sometimes I take off with just 8 or 9 satellites and it only increases slightly during the flight, whereas before—when I was using a Happymodel ES900TX/RX—I could easily get 15 satellites on the ground and even more during the flight.

Some info about my setup:

- Handset: Radiomaster TX16S 
- EdgeTX 2.10.5 (Centurion)
- TX: Radiomaster Nomad TX 2.4/900Hz
- RX: Radiomaster XR4
- RF Band: 868 EU
- TX Power: 500 mw
- RX Power: 100 mw
- Paket rate: K1000 Full(-101 dBm)
- Telem ratio: 1:2(39891bps)
- Switch mode: 16ch Rate/2. 
- Antenna mode: Gemini
- Link mode: Mavlink
- Flight Controller: Speedybee F405 Wing
- Flight Firmware: Arduplane 4.4.4
 

[image]

tguillaume02, 2025-04-11:
> J'ai un problème similaire avec le module GPS Walksnail lorsque j'utilise la bande 868 EU avec le Radiomaster XR4. Le temps d'attente pour atteindre même 8 satellites est très long. Il m'arrive de décoller avec seulement 8 ou 9 satellites et le temps d'attente n'augmente que légèrement pendant le vol, alors qu'avant, avec un Happymodel ES900TX/RX, je pouvais facilement capter 15 satellites au sol, et même plus pendant le vol.
> 
> Quelques informations sur ma configuration :
> 
> * Combiné : Radiomaster TX16S
> * EdgeTX 2.10.5 (Centurion)
> * Émetteur : Radiomaster Nomad TX 2,4/900 Hz
> * Récepteur : Radiomaster XR4
> * Bande RF : 868 EU
> * Puissance d'émission : 500 mW
> * Puissance de réception : 100 mW
> * Débit du paquet : K1000 Full (-101 dBm)
> * Rapport télémétrique : 1:2 (39 891 bps)
> * Mode de commutation : 16 canaux Rate/2.
> * Mode antenne : Gemini
> * Mode de liaison : Mavlink
> * Contrôleur de vol : Speedybee F405 Wing
> * Micrologiciel de vol : Arduplane 4.4.4
> 
> [image]

I tried today with 100hz low band I can get the GPS but if I switch to 2.4ghz I lose them.
I also have the compass moving all the time, if I check on the mission planner it's the GPS track back line that keeps moving.

a1axx, 2025-04-13:
> Same issue here. Nomad + DBR4. Rock solid sat count in 200 full low band, instant drop when switched to 333hz full 2.4G. Unscientific, but definitely confirms that there is a correlation between 2.4G and GPS loss of integrity.

I would suspect it’s something making noise on the receiver itself than the 2.4ghz carrier causing the problem.

I only see the issue on the dbr4 when switching frequencies, you might be getting the issue all the time if your supply voltage is different.

Another person raised it to Radiomaster too, they don’t seem to care, just said that nearby electronics could be affected by rf.

tguillaume02, 2025-04-22:
For those encountering this issue, do you have it in FCC or LBT?

chris1seto, 2025-04-22:
@tguillaume02 FCC, though on my end there was an interesting development. I now believe my GPS interference was coming from my flight computer. It coincided exactly with swapping to the DBR4, and I'm not sure what changed abruptly. I swapped to a NewBeeDrone F405 and suddenly the issue went away. All that to say: It's possible my report was errant.

a1axx, 2025-04-25:
> For those encountering this issue, do you have it in FCC or LBT?

Lbt for me

tguillaume02, 2025-04-25:
> > For those encountering this issue, do you have it in FCC or LBT?
> 
> Lbt for me

Also LBT for me, I haven't try with FCC

pkendall64, 2025-04-25:
Bear in mind that the LR1121 based devices can't do LBT yet.

tguillaume02, 2025-04-30:
I tested it in FCC, and the problem is still there, but less serious.
If I'm in full X100 mode, I can only get a maximum of 10 GPS during the flight - this sometimes drops to 0, compared with over 18 if I'm not in this mode.

[image]

[image]

raul-ortega, 2025-04-30:
This is the result in my last flight on 26th of April, after updating ExpressLRS firmware to 3.5.4 on TX (Nomad) and RX (XR4). The rest of conditions are exactly the same (868 EU, K1000, Mavlink...) of my previous post, except the place, a different location. 

[image]

NeariX67, 2025-05-20:
Experienced the same thing today. X100 Full, 1000mW, GX12 and XR4. Speedybee F405 Wing mini.
When i brought my TX near to my plane, i had 0 satellites. Setting output power to 10mW and pointing the TX away from my plane, satellites increased up to around 20.
I did not experience the same with my 7inch, also XR4 but a Skystars F7 HD Pro 4 FC.
Using 868 MHz EU Mode.

raul-ortega, 2025-06-11:
> I have a similar issue in the GPS module Walksnail when using the 868 EU band with Radiomaster XR4. The wait time to reach even 8 satellites is very long. Sometimes I take off with just 8 or 9 [...]

I'v solved my issue replacing the Walksnail GPS module with a BZ-251. It is providing up to 17 satellites before taking off, and up to 25 satellites during the flight. I get the 3D fix in less than 30 seconds.

a1axx, 2025-06-11:
> I'v solved my issue replacing the Walksnail GPS module with a BZ-251. It is providing up to 17 satellites before taking off, and up to 25 satellites during the flight. I get the 3D fix in less than [...]

Fitting a better GPS to overcome completely unnecessary noise floor from a receiver is a sticking plaster. 

If you have any solar effects etc on gps you may still find random gps loss

NeariX67, 2025-06-11:
After my research, it's the transmitter, not receiver.

a1axx, 2025-06-11:
> I just recorded a video of my analysis of this problem. Previously, I expected that any 2.4G packet rate would cause the gps to stop working because at any 2.4G packet rate there was noise on the 1575MHz band. Today I tested with a FC and a gps module connected to observe the number of satellites and noticed that only Xrossband packet rates caused a decrease in the number of satellites.
> 
> https://youtu.be/gIoeg1KCPm8 satellite count can be observed on the bottom right corner.

Okay - just tested this with a DBR4 first.

Started with 15+ sats and in 200 full low band
Switched to X150 full - lost all satellites
Switched to 333hz 2.4 full - regained 15+ satellites

Test setup is a matek f765wse, matek m10q, DBR4 rx, plenty of separation, it’s a fixed wing fuselage.

No motors, servos, vtx etc connected. Only the flight battery.

Next is an xr4

NeariX67, 2025-06-11:
Which confirms that it's not the receiver, but the transmitter. Gemini module or GX12 to be exact.

janmj, 2025-06-15:
I can confirm/repeat this issue.
New Beluga with: Atomrc 405 fc, Nomad module, DBR4 receiver, TX16S. Setup with X100 full, could not get a lock, only 2-5 sats. Came home, saw this thread and switched to Lowband 100 full, and immediately got 13+ sats inside, haven't tried outside yet.

tguillaume02, 2025-06-17:
@flyingbandit @janmj Are you in FCC or LBT ?
I was in LBT and had the same problems after switching to FCC the problem has almost disappeared.

flyingbandit, 2025-07-23:
Still have issues with this - narrowed it down to being intermittently 2.4 band but always Xrossband

IgnacioHR, 2025-08-25:
Going a bit deeper into the issue I explored the source code. Then, I found I can open this url while in captive portal with the receiver:

http://10.0.0.1/lr1121.html

and noticed the radio is using firmware version 0x0101. Here is a link to the driver source code in the latest version. It looks like the lr1121 chip can use the GNSS radio so maybe there is something in the chip configuration or some bug in the driver that might affect us.

https://github.com/Lora-net/SWDR001/blob/v2.5.2/README.md

Note the URL at the top also provides a mechanism to update the LR1121 driver!

IgnacioHR, 2025-08-25:
But version 0x0103 worked for me!

Here are steps to fix this issue:

1. Download LR1121 firmware version 0x0103 from here:
https://github.com/Lora-net/radio_firmware_images/blob/master/lr1121/transceiver/lr1121_transceiver_0103.bin

2. Connect your receiver using WIFI and head a browser to:
http://10.0.0.1/lr1121.html

3. Update firmware on both radios

4. Open the transmitter and configure it to X100 Full (in my case I tested this mode)
5. Power up your drone, wait for bind and satellites

This worked for me. Hope others can report the same too.

flyingbandit, 2025-08-27:
Will leave it open until I get a few test flights in. I was able to maintain sats indoors on x100. Just want to do a test flight :)

pkendall64, 2025-08-27:
Firmware 0104 is incompatible. We support it in 3.6.0 onwards. If it's doing the red-flash "bad radio" it will automatically start wifi mode. You should be able to go to the lr1121.html page and upload the 0103 firmware.
Another option is to flash the 3.x.x-maintenamce brach, which will become 3.6.0 and that automatically upgrades the LR1121 to version 0104.

ProfDrYoMan, 2025-08-28:
Just asking. Rough estimate on when to expect 3.6.0?

lesykm, 2025-08-30:
What if I experience this with SX1280 chip? Is there firmware with fix for it?

pkendall64, 2025-08-30:
> What if I experience this with SX1280 chip? Is there firmware with fix for it?

No, the SX1280 is not firmware upgrdable

flyingbandit, 2025-10-30:
Same here - further testing and it has showed itself to come back.

Ive found betaflight’s gui is a good GPS tool.

While the number of sats may not change in X band or 2.4 333, you can see lots of unused sats fleet in and out as well as the ground speed dart all over the place.

Higher powers and higher telemetry ratios (1:2 being the highest) increase the errors.

Low band has no GPS problems.

J-Sorenson, 2025-12-18:
I expect an increased telemetry rate to certainly cause an impact, as MavLink demands nearly 1:2 telemetry.  However, do I understand you correctly that a *lower* Rx power(10mW,25mW) is causing more *more* GPS interference than higher Rx telemetry power (50mW, 100mW)?  That seems backwards to me.

koriaf, 2026-03-04:
Some more stats:
* 50Hz
* Betafpv SuperX, 5cm from the HGLRC M100-5883 M10 GPS, with 1/64 telemetry: no issues, 30 satellites, seconds for the lock
* after increased telemetry rate to 1/2 (yes, mavlink) - 5-7 minutes for 3d lock (6 satellites), okay performance during the flight (slowly growing to 16 satellites) - visible degradation.
* moved GPS 50cm away from the RX antennas - back to 30 satellites, quick lock.
* both GPS and the receiver powered from the same 5v rail, no extra shielding.

Arthur-de-Partuur, 2026-08-21:
I am experiencing the same issue (really happy I stumbled upon this github issue as I have fully rewired my gps, replaced with others, split power and signal cables etc with no changes)
I am outfitting an 
- AtomRC Swordfish (fixed wing) with a 
- GEPRC M10 GPS (no compass) on a 
- iFlight Blitz H743 Wing running INAV 9.1
- Receiver is a RM XR4 (initially in X-Band, 100HZ full, 1/2 telemetry ratio, 16ch Rate/2 switch mode)
- TX = RM TX16S-Mk3 with ETX 2.12.1
- TX = RX on ELRS 4.1.0 ISM (yes, in Europe... I am a bit naughty)
- TX on Dynamic power, always at 10mW

Initial config was with the GPS about 5cm (~2 inches) away from the two antennas that were mounted on a 90 degree angle from each other in a horizontal plane. (ELRS is always in Dynamic mode at 10mW on the bench or during outside testing)
[image]
I never got any Satellites fixed. With u-center2 connected, I could see that the GPS saw satellites (up to 21 for GPS and lower numbers for Beido and Galileo) but they never fixed. 
When I finally figured out that this had to do with GemX and reverted to 2.4, I almost immediately got 4 satellites. Also switching to 868MHz gave me a fix.
I played around with the ELRS settings and saw that packet rate and even changing telemetry ratio seemed to have an influence on the nr. of sats it could fix.

I now have the antennas moved as far away as I can (in the battery bay) but they are now parallel to each other which is less than ideal (O4 air unit is not connected and was removed during tests).
[image]

Still I have 2 questions
1. What will be the best ELRS configuration
2. What is the performance going to do once in the air

Setting this up on 2.4GHz kind of negates the GemX receiver. I just as easily could switch to the XR3 or RP4 receiver (also a bit useless that I bought a TX16S-Mk3 to be able to have Gemini for LoRa...)

Also, I found that if I switch to (e.g.) 2.4GHz and 1/64 telem and I get a 7 GPS fix, I could go back to X-Band and keep some sats fixed (drops to 4 so not enough but in "city-environment so could improve when actually in a field) with my preferred settings.

Anyway, really curious on development of this and wanted to add some fixed wing experience in this topic.

Please let me know if I can help testing stuff in case somebody is working on this!

komodev, 2026-08-21:
I would definitely never place 04 or 03 vtx that near GPS especially on a higher position than GPS. That is the main problem for your GPS fix, much bigger than GemX protocol.
Try swapping your RX and 04 positions a bit. Put 04 on a front Hatch 04 mount and RX where your current 04 position is on second image. This will make GPS fix much faster, you'll see.
put your RX antennas below GPS aside from both sides and glue them, leave GPS there but find some mount or print it with at least 2cm height to put it a bit higher and your GPS fix will be faster and more reliable no matter if you use GemX or just 2.4g protocol.

Note: on RX position on second image: the antennas inside on both sides of a battery bracket will have a battery and XT670 discharge cable in between which is a current noise environment that is very bad for RX signal or anything placed that way. Move it on one of the wings or put it where 04 is now on second image.
All my comments and your equipment position is related to 2nd image. First is even bigger disaster radio-waves vise.

Arthur-de-Partuur, 2026-08-21:
Thanx for your quick response! Highly appreciated (longtime modelflyer but new to FPV). 
I don't want to Hi-Jack this thread too much as it is about the interference and not about my setup,... but still :-)
> I would definitely never place 04 or 03 vtx that near GPS especially on a higher position than GPS. That is the main problem for your GPS fix, much bigger than GemX protocol.

As stated in my post, O4 was never connected or placed on the model during my testing issues so we can rule that out for the GPS-lock issues. (I will take your suggestion into consideration, obviously, it is just not relevant to the issue)

>Try swapping your RX and 04 positions a bit. Put 04 on a front Hatch 04 mount

Yes, this was in the planning. Ordered a Pan/tilt unit to go on top of the Bat-hatch... Slow supply from China, though...
>and RX where your current 04 position is on second image. This will make GPS fix much faster, you'll see. 

But this would bring the ELRS antennas much closer to the GPS, right? That's why I now put them so much forward...
>put your RX antennas below GPS aside from both sides and glue them, leave GPS there but find some mount or print it with at least 2cm height to put it a bit higher and your GPS fix will be faster and more reliable no matter if you use GemX or just 2.4g protocol.

Thanx. I'll do that. I might mount it on top of the foam Hatch-cover to bring it up 3 cm. 
> Note: on RX position on second image: the antennas inside on both sides of a battery bracket will have a battery and XT670 discharge cable in between which is a current noise environment that is [...]

Sucks as I hotglued the whole thing in the bay allready,... well,... out she comes again... 

Thanx,... Back on topic ;-)

raphaelhunziker1202-stack, 2026-08-25:
For everyone following this: the fix for the X-Band GPS jamming discussed here is PR #3390 (FHSS GNSS avoidance). I posted a clause-level EN 300 328 analysis over there addressing the 'fewer channels breaks LBT compliance' concern (short version: a reduced hop set is fine — minimum is 15 frequencies at this channel spacing, and removing channels from a hopping sequence is explicitly permitted), and opened #3746: a rebase of #3390 onto current master (it had merge conflicts) plus a fix for a pairing-index bug found while verifying the sequence builder (~16% of EU868 X-Band hops were still landing intermod products inside the keep-out window; 0 after the fix). Bench/flight testers with X-Band hardware (XR4/GX12/Nomad + GPS) very welcome over on #3746.

flyingbandit, 2026-10-13:
> 29\. For everyone following this: the fix for the X-Band GPS jamming discussed here is PR [#3390](https://github.com/ExpressLRS/ExpressLRS/pull/3390) (FHSS GNSS avoidance). I posted a clause-level EN 300 328 analysis over there addressing the 'fewer channels breaks LBT compliance' concern (short version: a reduced hop set is fine — minimum is 15 frequencies at this channel spacing, and removing channels from a hopping sequence is explicitly permitted), and opened [#3746](https://github.com/ExpressLRS/ExpressLRS/pull/3746): a rebase of [#3390](https://github.com/ExpressLRS/ExpressLRS/pull/3390) onto current master (it had merge conflicts) plus a fix for a pairing-index bug found while verifying the sequence builder (~16% of EU868 X-Band hops were still landing intermod products inside the keep-out window; 0 after the fix). Bench/flight testers with X-Band hardware (XR4/GX12/Nomad + GPS) very welcome over on [#3746](https://github.com/ExpressLRS/ExpressLRS/pull/3746).

There is still a problem in 2.4 modes (at least 333Hz) - I have been reproducing the behaviour across several gps. The only way to get GPS performance is to turn the TX off and lose the bind to the RX (still powered on).

I was getting 15-20 sats with the tx off and 0-9 with the tx on.

20+cm of separation. Tried with several different antennas.

Updated the OP