ExpressLRS/ExpressLRS #3187: RX disconnects and reconnects approximately every 10 seconds

ama9910, 2025-04-15:
## Current Behavior

RX disconnects and reconnects approximately every 10 seconds. Tx will issue warnings such as "Telemetry Lost", "RF Signal Critical", "Telemetry Recovered" over and over. ExpressLRS Lua script shows "C" in top right corner for a few seconds then "-" when connection is lost, then back to "C".

## Steps to Reproduce

I can only reproduce on two new RXs that I'm trying to configure. Several existing RXs (all of these are same model) do not exhibit the problem. I suspect if I try and update the existing (working) ones, the problem will start occurring on those too. I have successfully updated the older RXs before. This issue has only started occurring recently.

## Details

DEBUG_LOG:
New TLMrate 1:16
tentative conn
Bad sync, aborting
lost conn fc=0 fo=0
Config LoRa tentative conn
Bad sync, aborting
lost conn fc=0 fo=0
Config LoRa tentative conn
Bad sync, aborting
lost conn fc=0 fo=0
Config LoRa tentative conn
Bad sync, aborting
lost conn fc=0 fo=0
Config LoRa tentative conn
Bad sync, aborting
lost conn fc=0 fo=0
Config LoRa tentative conn
Bad sync, aborting
lost conn fc=0 fo=0
Config LoRa tentative conn
Bad sync, aborting
lost conn fc=0 fo=0
Config LoRa 

DEBUG_LOG_VERBOSE will produce a few extra lines:
CRC error
MM 35=35 1
as well as the verbose stuff:
-538:-2150:-135:0:33
-1114:-2843:-170:0:50
-1295:-1837:-96:0:60
-1683:-2847:-154:0:66
-1722:-1836:-81:0:71
-2000:-2834:-138:0:75
-596:3616:274:0:77
811:5034:345:0:80
149:-1837:-106:0:81
-599:-2842:-162:0:83
-907:-1830:-89:0:84
-1391:-2842:-147:0:85
-52:3966:288:0:86
1534:6290:415:0:87
439:-2843:-182:0:88
-130:-1837:-107:0:88
-809:-2847:-164:0:89
-1066:-1836:-91:0:90
etc

I have tried 
- ELRS 3.5.4, 3.3.2, 3.2.0
- EdgeTX 2.10.6, 2.6
- Baudrate 1.87M, 400k
- Packet Rate 50Hz, 250Hz (no connection at all at 250Hz)
- Rx powered by BEC, ESC, FTDI cable
- Another Rx of same type
- "Clear PlatformIO Dependancies" and "Clear Firmware Files" in Configurator
- "Erase Before Flash" in Configurator
- "Reset all model settings to defaults" in Rx web page
- Programming via Wifi and FTDI/Serial

All combinations exhibiting the same issue
Rx is on bench with 5v supplied, no servos or FC connected

## Your Environment

- TX hardware: Radiomaster Tx16s, internal ELRS Tx
- RX hardware: Matek R24P-6 PWM Rx
- EdgeTX version 2.10.6
- ExpressLRS version (TX & RX MUST MATCH): 3.5.4 on both
- Packet Rate: 50Hz
- Telemetry Ratio: 1:16
- user_defines:
  -DMY_BINDING_PHRASE="noworky"
  -DRegulatory_Domain_ISM_2400
  -DHOME_WIFI_SSID="redacted"
  -DHOME_WIFI_PASSWORD="redacted"
  -DAUTO_WIFI_ON_INTERVAL="60"
  -DLOCK_ON_FIRST_CONNECTION
  -DDEBUG_LOG

mha1, 2025-04-15:
Because I believe this is one of the unsupported Matek receivers I wonder which target are you flashing?

Paulraaf, 2025-04-15:
Are you building the firmware yourself?  I once had a similar issue where I was using the Unified_ESP8283_2400_RX target with the "Matak R24-P PWM 2.4Ghz RX" configuration applied after building. This configuration is incorrect for the Matek R24-P6. I believe "Generic ESP8285 6xPWM 2.4Ghz RX" was the correct one.

ama9910, 2025-04-16:
Argh. I had no idea that [r24-p](https://www.mateksys.com/?portfolio=elrs-r24-p) and [r24-p6](https://www.mateksys.com/?portfolio=elrs-r24-p6) were different things. I was trying to flash with the wrong target selected. I've now used the "HappyModel EPW6 2.4GHz PWM RX" target (as Matek suggest on their p6 web page) and it seems to work fine. Thanks for the comments.

closed 2025-04-16 by ama9910