ExpressLRS/ExpressLRS #3195: Betaflight ExpressLRS SPI aux channel decode error

Fourflies, 2025-04-26:
## Current Behavior

In Betaflight, Aux channels get decoded to incorrect values. It looks like an integer overflow or an MSB getting chopped in the channel decoder. When the transmitter sends in the range 1000-1500 µs it gets decoded in Betaflight to a range of 1000-2000 µs. Just as the transmitter goes over 1500 µs, the Betaflight decode jumps back to 1000 µs and then the transmitter's 1500-2000 µs range gets decoded as 1000-2000 µs in Betaflight. Only occurs on the Aux channels. Channels 1-4 are unaffected.

## Steps to Reproduce

Specific to ExpressLRS SPI in Betaflight. Unable to reproduce with a serial ELRS receiver. Confirmed in Betaflight 4.5.2, 4.5.1, and 4.4.3.

Narrowed down to an interaction between Wide switch mode and telemetry. In Hybrid switch mode, channel decodes work correctly regardless of telemetry. In Wide switch mode, if telemetry is disabled in the TX module, the Aux channel decode bug occurs. It seems to affect all Aux channels. Enabling any telemetry ratio in the TX module fixes the issue.

## Possible Solution (Not obligatory)

Wild guess, but maybe something in this function:

https://github.com/betaflight/betaflight/blob/master/src/main/rx/expresslrs.c#L271

## Details

## Your Environment

- TX hardware: Radiomaster Pocket internal module
- RX hardware: HDZero AIO5 flight controller
- Handset model: Radiomaster Pocket
- OpenTX version (including nightly number): 
- ExpressLRS version (TX & RX MUST MATCH): 3.3.1/3.4.3 and Betaflight 4.5.2/4.5.1/4.4.3
- Packet Rate: 50 Hz
- Telemetry Ratio: multiple
- user_defines:

pkendall64, 2025-04-26:
This bug should be raised in the Betaflight issue tracker. The ELRS team does not maintain the ELRS implementation in Betaflight.

closed 2025-04-26 by pkendall64

Fourflies, 2025-04-26:
@pkendall64 Done, thanks! In case anyone looks for it - https://github.com/betaflight/betaflight/issues/14369