ExpressLRS/ExpressLRS #3199: Not enough memory to run elrsV3.lua with x9dp (stm32f2) running EdgeTX 2.11.0

slackista, 2025-05-04:
## Current Behavior

Unable to run elrsV3.lua on x9d+ with stm32f2 mcu. All common scripts like betaflight are working fine.

## Steps to Reproduce

1. Update radio EdgeTX 2.10.5 → 2.11.0
2. Remove all .luac compiled files from sd
3. Execute elrsV3.lua
4. Error, not enough memory

## Details

In v2.11.0 EdgeTX they moved to lua 5.3. That is breaking change https://github.com/EdgeTX/edgetx/pull/4203
In v2.10.5 while elrsV3.lua is running I've got  GV Use: ~45kb

J-Sorenson, 2025-05-14:
Does it generate the compiled script?  It may work after a transmitter reboot.  On my X-Lite with an earlier version of DSM forward programming, I would have to first attempt to use the script, but it only had enough memory to compile it.  I would then restart my transmitter (clear memory), and then it would run from the compiled script.

Alternatively, (I haven't tested this yet), I thought they were going to add the ability for the Companion software to compile the script.

slackista, 2025-05-16:
Confirm, compiling via Companion and run on radio works. Fresh startup radio and run .lua doesn't.

pkendall64, 2025-05-16:
Closing as this is not an ELRS issue. The issue is that these older radios just don't have enough memory for the Lua compiler to build some more complex scripts.

closed 2025-05-16 by pkendall64