ExpressLRS/ExpressLRS #3207: Wide Switch AUX channel 'glitches' with Rotorflight

breavyn, 2025-05-07:
## Current Behavior

Occasionally, AUX channel output flutters rapidly for several seconds before returning to normal.

## Steps to Reproduce

I've been able to reproduce the issue by creating a very poor link quality and enabling the Rotorflight background script in EdgeTX. This script sends several MSP messages back and forth when it connects to a model. A connect is when RSSI transitions from 0 to non 0.

It does not seem to occur when this script is disabled, i.e. only using RC packets and Rotorflight custom telemetry packets.

## Your Environment

- TX hardware: RadioMaster TX16S internal 2.4G
- RX hardware: RadioMaster RP3-H
- Handset model: RadioMaster TX16S
- EdgeTX 2.11.0
- ExpressLRS version: 3.5.4
- Packet Rate: LoRa 500 and F1000
- Telemetry Ratio: 1:8
- RadioMaster Nexus w. Rotorflight 2.2-RC3

breavyn, 2025-05-08:
After some more testing, I have a theory as to what is causing this behaviour.

When a MSP message is sent, the TX changes the telemetry ratio to 1:2. If the prior telemetry ratio is 1:8 or slower, this will cause the resolution of the AUX channels to go from 7 bit to 6 bit. The problem arises when the RX does not successfully receive the sync packet. This leads to the TX sending 6 bit values, and the RX trying to read them as 7 bit.

I tried using a telemetry ratio of 1:4, which uses 6 bit AUX resolution, and found I was unable to replicate the issue.

Further, I added the following debug logging to the RX firmware, to show the OTA values for the first AUX channel.

```diff
diff --git a/src/lib/OTA/OTA.cpp b/src/lib/OTA/OTA.cpp
index 219b9941..775bd1ee 100644
--- a/src/lib/OTA/OTA.cpp
+++ b/src/lib/OTA/OTA.cpp
@@ -421,6 +421,10 @@ bool ICACHE_RAM_ATTR UnpackChannelDataHybridWide(OTA_Packet_s const * const otaP
             switchValue = switchByte & 0b1111111; // 7-bit
         }
 
+        if (switchIndex == 0) {
+            DBGLN("| AUX1: %u / %u", switchValue, bins);
+        }
+
         channelData[5 + switchIndex] = N_to_CRSF(switchValue, bins);
     }
 
```

Adjusting the channel output to be `91 / 127`, then triggering the issue, leads to random readings of `45 / 127` and `109 / 127`. The 127 indicates the RX is not swapping to the 1:2 telemetry ratio.

45 and 109 are the two permutations of the 7th bit when the 6 bit value is 45. `45 / 63` is the same CRSF value as `91 / 127`.

CapnBry, 2025-05-09:
Wow this is a great diagnosis and 100% what is happening. The two systems were developed so far apart in time that I didn't even think about how it would interact if the syncspam for the ratio change didn't get there.

Also I never thought that people would ever be so silly to do data transfer while in flight due to how it severely impacts the uplink rate (i.e. the sticks being sent at half rate).

Mickey's got a solution coming though, thanks for figuring this out!

offer-shmuely, 2025-05-09:
Well, looks like your talking about me doing the "silly" msp communication during flight 😀😎

I am interesting to know why I did not feel it during flight? Can it be that it is not happening in 16/2 mode? in any tlm rate?

pkendall64, 2025-05-09:
> Well, looks like your talking about me doing the "silly" msp communication during flight 😀😎
> 
> I am interesting to know why I did not feel it during flight? Can it be that it is not happening in 16/2 mode? in any tlm rate?

The issue only affects wide switch mode.

CapnBry, 2025-05-09:
The channel value changing only affects Wide switch mode, but the tlm ratio changing does affect everyone. If you're running 100Hz Full 16 channel mode, that's 50Hz. The instant you want to transfer MSP, the link drops to 50Hz symmetrical, or 25Hz of channels/control packets. If it is 333Hz Full, then it goes to 166.5Hz in 16ch mode, or 83.25Hz of channels packets when MSP is queued on the TX.

offer-shmuely, 2025-05-10:
What is the difference between the first 50hz, to the second one, the symmetrical?
25hz for channels + 25hz for msp?

CapnBry, 2025-05-10:
Sort of. The over-the-air bandwidth is split between uplink and downlink by choosing a telemetry ratio, e.g. for 100Hz the default is 1:32 or 1 out of every 32 packets is downlink (telemetry) and the other 31 are uplink (channels). That means that the channels link is effectively 96.875Hz.

When MSP data is sent to the receiver, the link automatically goes to 1:2 mode, meaning 1 packet up for every 1 packet down, or "50Hz symmetrical". Then you divide that in half again because half of the uplink packets are CH1-8 and half are CH9-16 leaving you with a 25Hz channels uplink. To subtract from that even further, because you're sending MSP data up to the RX, some of those channels packets are now going to be MSP data instead of channels making the control link even lower than 25Hz.

So that's why it isn't great to send MSP data in flight if you want responsive control because your control link could theoretically drop as low 12.5 updates per second if there's always MSP data to send to the RX.

offer-shmuely, 2025-05-10:
Thanks for detailed explanation, now I feel I better understand what is the exact meaning of the tlm rate 😀😀

This calculations assume that all the time there is an msp calls waiting in the queue.
However, In my widgets implementation, I do not send them all the time.
I do about 4 msp calls each second when not arm.
and when armed (both elrs & rotorflight) I do the 4 msp calls once in 5 seconds.
I need to find out  how it really affects the heli handling.

CapnBry, 2025-05-10:
Speaking of 5 seconds, that's how long this Burst mode typically lasts IIRC. When we need to send a sync packet (which occur on the first time the frequency hopping is on the "sync" channel after every 5 seconds) if there's MSP queued, it will request the other side go to the faster telemetry speed and will stay at that rate until the next sync packet.

So basically 4 MSP calls every 5 seconds will keep the link in "burst" transfer mode constantly.

Junnicutt, 2025-06-07:
Would using the betaflight lua scripts background script trigger this as well? The github page mentions it uses  MSP for RTC sync and RSSI. But RSSI only if no other rssi source is available. 
I'll probably set the lua script to only be active when disarmed just to be safe.

CapnBry, 2025-06-08:
The Betaflight lua does appear to send RTC information when the model connects and trigger the Telem Boost. It doesn't appear to have any activity after that. The background script's only purpose appears to be to set the clock on the FC, which I am not sure why that would be needed when that information can come from the GPS. I would just say to not use the background script at all, since it does not affect the ability to use any of the other BF lua functionality.

The inav lua script used to just be a way to display telemetry, I do not believe it does any send/receive. Not sure about Yaapu at all.

I don't believe we're even aware of the full compliment of what scripts users might run and can't fully test them all to determine if they **never** send data while in flight. Users will have to ask the authors of the script they want to use if it sends data while in flight to allow them to make their own decision if that's important enough to keep the link in boost mode.

Junnicutt, 2025-06-09:
Yeah that makes sense. Is there a blackbox debug mode or anything we could use to check for issues or changes due to a script?
I used the background script solely for the rtc feature. It was great for getting a time and date on analog osd on toothpicks and whoops. But yeah its no big deal to just turn it off to be safe.

CapnBry, 2025-06-09:
You can use RC_SMOOTHING_RATE maybe to tell, but betaflight is pretty good about figuring out the packet rate even with telemetry in the mix. The first number is uS between packets divided by 10 (e.g. 250Hz / 4ms would show as 400, 500Hz 2ms = 200). The third number is what it thinks the packet rate is (e.g. 250 should be 249 or 250).

In 1:2 boost mode, it is very clear that the first / third numbers are double / half what you'd expect them to be respectively. However now that Wide only boosts to 1:8, the third number doesn't change and the first number just blinks to double and the third stays at the correct packet rate. If you're only testing a script to see if it is safe, you can run Hybrid switch mode (which does boost to 1:2) and then it should be clear since the numbers are more consistent.

Users with OLED/LCD screens on their TX module will see the current actual telemetry ratio so for them it is fairly obvious.

PR #3210 (2025-05-10): TLM ratio boost: don't cross the wide switch 7-bit to 6-bit boundary  
fixes #3207

Thanks to @breavyn for the detailed issue and excellent analysis
Thanks to @CapnBry  for the always great inputs and discussions