ExpressLRS/ExpressLRS #3209: Bugs found on ExpressLRS RX with Gyro Support

CarlouT, 2025-05-07:
The attached dpocument below details the steps I followed. In eah chapter i have also documentd the bugs I found .  Here's a summary of the bugs I found during my testing.
[How to configure the GYRO.pdf](https://github.com/user-attachments/files/20094424/How.to.configure.the.GYRO.pdf)
1.	With the BetaFPV SuperP receiver, after flashing the GYRO firmware, it does not enter WiFi mode even after the WiFi Interval time has elapsed. Refer to Chapter 5 for a workaround. I have not tested other receivers.
2.	The Gyro Align on FLIP (Receiver inverted) are not working. Refer to chapter 13 for more details.
3.	The master gain is only applicable to Rate mode. You will not be able to remotely adjust the gain on the other modes (i.e. SAFE, LEVEL HOVER). On these modes, the max gain set in the Axis gain is used. If its too much, the plane will shake on that axis during flight. The only way to fix it is to land and adjust the axis gain of that surface. Refer to chapter 14 for more details.
4.	Can I request to add 5 positions in the mode channel in order to control the five modes if the GYRO either via two switches or via the 6-position switch of the RadioMaster? Refer to Chapter 11

- TX hardware: Radiomaster TS16S
- RX hardware: BetaFPV SuperP 14CH ELRS REceiver
- ExpressLRS version (TX: edgetx-ts16s VERS: v2.10.6 & RX; downloaded firmware from https://github.com/awigen/ExpressLRS/tree/add-rx-gyro-support
- Packet Rate: 333Hz
- Telemetry Ratio: 1:4  (6549bps)
- user_defines: -DHAS_GYRO and -DGYRO_DEVICE_MPU6050

I'm happy to share more info if needed. I am also happy to do more testing. I have a foamy plane that I will be using to test this GYRO.

secretman28, 2025-05-08:
agreed but i think you need to make the request on the awigen fork and not official elrs, im working on this setup too and your right bug or not all implanted completely for now still search if the current official elrs have the gyro part in it add gyro fork has update fromg a long time i dont think he still work on this there im hopping official elrs switch

CarlouT, 2025-05-09:
Transfered to awigen as suggested by secretman.

secretman28, 2025-05-09:
still this should be implented in main elrs this will help a lot to get airplane/jet guy`s to elrs

CarlouT, 2025-05-09:
Ok re-opening this issue in the main ELRS. Everybody is looking forward for this feature to be completed.