ExpressLRS/ExpressLRS #3256: ImmersionRC Ghost Atto / ELRS 3.5.6 - RX does not enter binding mode after power cycling

SnowLeopardFPV, 2025-06-20:
## Current Behavior

After flashing my Ghost Atto with ELRS 3.5.6 firmware (via Betaflight Passthrough) it reboots and the LED starts flashing yellow to indicate that it is in binding mode. When the TX is powered up it successfully binds to the Atto with the LED turning solid blue (250Hz packet rate). If the TX is powered back off the Atto enters binding mode again with a flashing yellow LED. This process can be repeated successfully as long as the Atto isn't power cycled since the ELRS 3.5.6 firmware was flashed to it.

After power cycling the Atto, the RGB LED on it initially turns a solid greeny-yellow colour, then it turns a solid cyan colour and gets stuck there. There is no rainbow cycling and it never flashes yellow, so there is no way to bind to it. No amount of power cycling resolves the issue. Reflashing ELRS 3.5.6 firmware to it will allow binding again as described in the first paragraph above, but as soon as the Atto is power cycled it gets stuck on the solid cyan LED again as described in this paragraph.

If the Atto is flashed back to ELRS 3.5.5 everything behaves as it should and it binds to a TX with no issues even after power cycling the Atto, so the issue is something that has been introduced in the ELRS 3.5.6 firmware.

## Steps to Reproduce

Flash ELRS 3.5.6 firmware to a Ghost Atto RX and try what is described in the Current Behaviour section above.

## Possible Solution (Not obligatory)

This might be the same or a similar issue to that of a previously reported and fixed issue (#3085) and could be due to lack of memory in the STM32 based receivers, so it's possible that it may affect other STM32 receivers and not just the Ghost Atto, but I have no other STM32 receivers to test ELRS 3.5.6 on.

## Details

There are no additional details. Everything of relevance is described in the Current Behaviour section above.

## Your Environment

- TX hardware: ImmersionRC Ghost Atto 2.4GHz
- RX hardware: NamimnoRC Flash OLED 2.4GHz & Radiomaster Nomad
- Handset model: FrSky Taranis X9D+ SE (ACCST version)
- EdgeTX version: 2.10.6
- ExpressLRS version (TX & RX MUST MATCH): 3.5.6
- Packet Rate: 250Hz
- Telemetry Ratio: Std (1:64)
- user_defines: N/A

123GoqkadMe, 2025-06-25:
RX does not enter binding mode after power cycling #3256

pkendall64, 2025-11-05:
The issue is that the Atto is running out of flash and the fake EEPROM space is erasing some of data block which is causing the issue. The only way to get this fixed is to remove more features!
I can remove SUMD and DJI RS Pro protocols and it fits, i.e. can power off/on the RX multiple times and it reconnects to the transmitter without issues. I can also get/modify the parameters for the RX via the Lua script as well.
Could you try PR #3386 and let me know.

SnowLeopardFPV, 2025-11-05:
@pkendall64 I flashed PR #3386 to my Atto and can confirm it now works as expected, and it successfully binds to a transmitter that is running one of ELRS 3.5.5, 3.5.6 or 3.6.0 (I haven't tested it with earlier 3.x versions but I'm sure it's probably fine with those as well). I can also successfully change parameters on the Atto via the ELRS LUA script. So removing those two less used protocols from the Atto firmware appears to have done the trick.

I really appreciate the time and effort you have spent on this, so thank you for looking into ways to resolve the issue. I realise at some point I'm going to have to bite the bullet and replace the Atto on that quad with a newer ESP32 based one if I want to keep up-to-date with any new ELRS releases and features, especially once ELRS 4.x gets released where STM32 support is being dropped, but if it is possible to keep the older STM32 based receivers working on the ELRS 3.x.x releases then I think that is a good compromise. Hopefully no-one is using a Ghost Atto with either the SUMD or DJI RS Pro protocols.

closed 2025-11-22 by pkendall64